safepay/sensor.fronius #2: Other Sensors

cathelest, 2019-07-13:
Sorry not sure if right place to post,

Love the new add on many thanks, much cleaner and better implementation then what i am using,

I use 3 other sensors which are not in the addon on i think?
Although i also have the fronius smart meter, so not sure if these are available to others without it?

These are 

P_Grid which will give me a negative value if sending power to grid, or negative value if drawing from grid.

P_Load - Must be a smart meter one i guess?
P_PV - Shows amount of energy being produce by panels? None of the plugin ones seem to match that one so not sure if i am missing it?

Thanks
Ben

This is the below i have been using, I had a sensor 1, which was day total i have removed as was in the plugin,

#FRONIUS GRID USAGE NEGATIVE VALUE IS SENDING POWER BACK POSITIVE IS DRAWING FROM GRID
sensor 2:   
    platform: rest
    resource: http://**IPOFINVERTER**/solar_api/v1/GetPowerFlowRealtimeData.fcgi
    name: Fronius Grid Usage
    value_template: '{{ (value_json.Body.Data.Site.P_Grid ) | round(3) }}'
    unit_of_measurement: W
#FRONIUS LOAD
sensor 3:   
    platform: rest
    resource: http://**IPOFINVERTER**/solar_api/v1/GetPowerFlowRealtimeData.fcgi
    name: Fronius House Load
    value_template: '{{ (value_json.Body.Data.Site.P_Load ) | round(3) | replace("-","") }}'
    unit_of_measurement: W

#FRONIUS PANEL GENERATION CURRENTLY
sensor 4:   
    platform: rest
    resource: http://**IPOFINVERTER**/solar_api/v1/GetPowerFlowRealtimeData.fcgi
    name: Fronius Panel Status
    value_template: '{{ (value_json.Body.Data.Site.P_PV ) | round(3) | replace("-","") }}'
    unit_of_measurement: W

safepay, 2019-07-18:
Yes. This was missing as I started this project focusing on calling the Inverter URL (see README).

However, it seems a number of people also have the PowerFlow Meter.

I have updated the code (but not yet documented it) to append PowerFlow data.

Please see the master branch.

You will need to set "powerflow: True" in configuration.yaml

If this works, I'll create a new release.

Please file any issues.

safepay, 2019-07-18:
Component updated.

closed 2019-07-18 by safepay