safepay/sensor.fronius #4: Scanning Interval below 300s

mattclar, 2019-07-22:
Love your work!
But you've set a hard minium scan_interval in your code 
` MIN_TIME_BETWEEN_UPDATES = timedelta(seconds=300)`

not sure if this was on purpose but users trying to set `scan_interval` manually below this level are unable

safepay, 2019-07-23:
Thanks Matt.

What interval would you suggest?

I was aiming for an interval that did not put too much overhead on the network.

cathelest, 2019-07-25:
I would prefer to set my own, and would like 20 seconds min to try and mimic the frequency of updates on the fronius app, i likely will set higher but to play with myself would be great,

Also regarding overnight shutdown, the inverter does but the smart meter doesnt, i am away currently so running 7.0, will have a look at the code for the current when im home and let you know how i find it,

Many thanks again for taking the time to support and code it, i really appreciate it,

safepay, 2019-07-25:
Thanks for the nice comment.

To have a play, just edit sensor.py and change the the line:

``MIN_TIME_BETWEEN_UPDATES = timedelta(seconds=300)``

 to some thing else. E.g.

``MIN_TIME_BETWEEN_UPDATES = timedelta(seconds=20)``

That way you can test without me changing the code for everyone.

mattclar, 2019-07-26:
I was more pointing out that this  MIN_TIME_BETWEEN_UPDATES conflicts with
the ability to set  scan_interval to anything below 300sec.
Perhaps you should set a default scan interval of 60 sec and then set a
MIN_TIME_BETWEEN_UPDATES  of something like 5-10 seconds or so.
Good idea to be light on the network but I doubt that these json requests
are stressful on the network at all

Matthew Clarke
Dentist

Toorak Dental Group
(03) 9826 1337
1A Grange Rd
Toorak 3142 VIC
http://toorakdentalgroup.com.au

safepay, 2019-07-26:
scan_interval is inbuilt to HA and defaults to 30 seconds, but can be overriden for any entity by passing the scan_interval setting (I added it to the instructions for information)

See https://www.home-assistant.io/docs/configuration/platform_options/

MIN_TIME_BETWEEN_UPDATES (in the code)  is to set a floor limit for scanning, in case the scan_interval is too much for the end point to handle. I chose 300 seconds as a starting point that seemed reasonable.

E.g. MIN_TIME_BETWEEN_UPDATES  = 300
Default scan_interval = 30
So the API can only be scanned for a minimum of 300 seconds.
Setting scan_interval: 10 will be ignored in this case.
Setting scan_interval: 310 will set the API polling to 310 seconds.

I am happy to set the MIN_TIME_BETWEEN_UPDATES to anything (or even remove it) so long as the inverter is Ok.

So, for example, you can edit the code in the custom_components folder and change 300 seconds to 5 seconds.
Then the system will use HA's default scan_interval of 30 seconds, which you could then manipulate down to 5 seconds or increase to whatever suits.

safepay, 2019-07-27:
I've set the minimum scan_interval (via MIN_TIME_BETWEEN_UPDATES) to be 60 seconds.
This seems safe to me as a default.
If anyone has another solution, please let me know.

closed 2019-07-27 by safepay