safepay/sensor.fronius #6: Not An Issue!

cathelest, 2019-08-13:
Hello Richard,

Many thanks for all your work on the plugin, its great, and thanks for allowing the flowmeter to monitor after sunset, works great now, iv sent you some beer money, have 1 or 2 on me, enjoy!!

Do you have any plans to make a custom card view (like the fronius app) or do you know if theres a way to use the power wheel card https://community.home-assistant.io/t/lovelace-power-wheel-card/82374

Iv tried to use it before but could never work it out properly,

All im missing now is the view!!

Many thanks again for all your hard work, sorry to open an issue but dont know how to contact you otherwise??

Ben (Cathelest)

safepay, 2019-08-13:
Cheers Ben.Glad I could help and thanks for the beer money.

I'll look in to the power wheel when I have time.

Richard

safepay, 2019-08-15:
Hi Ben,

Not sure if this looks right on your system, but I configured the basic custom card like this:

```yaml
type: 'custom:power-wheel-card'
solar_power_entity: sensor.fronius_panel_status
grid_power_consumption_entity: sensor.fronius_grid_usage
grid_power_production_entity: sensor.fronius_grid_usage
```

Being sure to add this to the top of your Lovelace config in the Raw Config Editor.
```yaml
resources:
  - type: module
    url: /local/custom_ui/power-wheel-card.js?v=1
```

Note that I put the JS file in ../www/custom_ui/ folder in the HA config folder which is different from the instructions but suits my setup.

cathelest, 2019-08-15:
grid_power_consumption_entity: sensor.fronius_grid_usage
grid_power_production_entity: sensor.fronius_grid_usage

Same entity?

Iv attached pics of both the powerwheel card and fronius android app, if you can help great, but i know this isnt your project, so if not many thanks anyway, really appreciate your time,

[https://cathelest.com/sa.png](url)
[https://cathelest.com/sf.jpg](url)

Ben

safepay, 2019-08-15:
Yeah, copy and paste is not always your  friend.

I've edited the original post.

a-wolter, 2019-09-15:
For me, this here was working:
```
        title: Solar Power
        production_is_positive: false
        solar_power_entity: sensor.fronius_panel_status
        grid_power_entity: sensor.fronius_grid_usage
        home_energy_entity: sensor.fronius_house_load
```

Here you can compare the original fronius solarweb view to the result in home-assistant:

[image]

cathelest, 2019-09-15:
Many thanks A-Wolter, that worked perfectly, thanks heaps!

safepay, 2019-09-15:
Thanks. I've added this  as instructions in my README in case someone else wants to do this.

closed 2019-09-15 by safepay