safepay/sensor.fronius #7: Sensors Unknown 0.8.4 >

cathelest, 2019-09-02:
Hello,

Sorry to be a pain,

Just noticed that something breaks my setup since 0.8.4 >

My config

sensor:
  - platform: fronius_inverter
    ip_address: 192.168.1.75
    device_id: 1
    scan_interval: 20
    powerflow: True

0.8.3 Seems to work great,
Screenshot
[image]

0.8.4 > (Tested 0.8.4, 0.8.5, 0.8.6)
[image]

Thanks again for taking time develop this, 

Ben

safepay, 2019-09-14:
It's probably just not getting values when HA is started/restarted after sunset.

When the inverter data is restored the next day after sunrise, the values should populate.

I probably need to have some sort of "1st run" detection and insert zero's.

bosbefok, 2019-11-04:
I've noticed this same issue as well. No values are returned if the system isn't generating AC power. I would have thought at least the energy values should be there. Definitely setting some defaults would be a good idea. Cheers

cathelest, 2019-11-05:
Iv jumped back to 0.8.3 from memory and works fine, not sure what changed after that, can u jump back and see aswell?

safepay, 2019-11-05:
The change was adding the pause overnight to stop polling. This was to remove errors in the logs.

You can work around this by overriding the pause timing using the start and stop times parameters.

I am unsure of the exact nature of the problem, though. There is not enough detail.

What does "at least the energy values should be there" mean? Which specific values are those?

bosbefok, 2019-11-06:
I was referring to sensor.fronius_day_energy, sensor.fronius_year_energy, sensor.fronius_total_energy. All of which were returning unknown if the hassio instance was restarted after there was no AC production. I haven't tried 0.8.7 yet, but will update shortly and let you know of any issues I encounter.

Such a useful addon, thanks heaps for your effort.

bosbefok, 2019-11-07:
0.8.7 is running great. Haven't noticed any issues. Would you be able to add a units parameter for the power to allow the user to choose between W and kW so they don't need to setup template sensors. Cheers

safepay, 2019-11-15:
Which power sensors do you specifically want to have converted?

Would they be individually converted or is there a single conversion that you would apply to several of them?

bosbefok, 2019-11-16:
> Which power sensors do you specifically want to have converted?
> 
> Would they be individually converted or is there a single conversion that you would apply to several of them?

sensor.fronius_ac_power, sensor.fronius_grid_usage, sensor.fronius_house_load.
I'd just like the option to show these 3 in kW instead of W.

safepay, 2019-11-16:
> sensor.fronius_ac_power, sensor.fronius_grid_usage, sensor.fronius_house_load.
> I'd just like the option to show these 3 in kW instead of W.

I've uploaded a new sensors.py file to the master branch for you to try.

If this works for you, I'll do a release.

Cheers.

safepay, 2019-11-27:
This issues seems to be resolved

closed 2019-11-27 by safepay