safepay/sensor.fronius #8: 0.8.7 Sensors Missing

cathelest, 2019-11-12:
Using the same config on both 0.8.3 and 0.8.7
some of the sensors arnt being created,

Im missing 
sensor.fronius_ac_current
sensor.fronius_ac_frequency
sensor.fronius_ac_power
sensor.fronius_ac_voltage
sensor.fronius_panel_status

Config im using, (i havent changed anything between the 2 versions, was i meant to?)

Sorry to bother you with this again,
Thanks for your help,

Ben,

P.S in the home assistant HACS addon, the dropdown with the versions, all other versions have a 
v in front, eg v0.8.3, except 0.8.7, causing it to show at bottom of dropdown, not an issue but thought ill let you know,

# Sensors
sensor:
  - platform: fronius_inverter
    ip_address: 192.168.1.75
    device_id: 1
    scan_interval: 20
    powerflow: True

safepay, 2019-11-15:
No change to any of that code.

Do you have a log?

safepay, 2019-12-05:
No activity on this so closing.
If problem persists, please reopen.

closed 2019-12-05 by safepay