safepay/sensor.fronius #9: Update for sensor.fronius_ac_power fails after sunset

jaganath-b, 2019-11-22:
Following error message fills up the log every minute until the Inverter is back up in the morning

UnboundLocalError: local variable 'state' referenced before assignment
2019-11-23 04:50:09 ERROR (MainThread) [homeassistant.helpers.entity] Update for sensor.fronius_ac_power fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 270, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 448, in async_device_update
    await self.async_update()
  File "/config/custom_components/fronius_inverter/sensor.py", line 207, in async_update
    self._state = round(state, 2)

safepay, 2019-11-24:
I've tried fixing this but have not been able tto do so without breaking the overall usability. I'm looking for someone else to code hat for me and submit a pull request.

nilrog, 2019-11-24:
I have noticed the same thing...and also that the powerflow sensors are unavailable which makes the power-wheel-card show red errors about them. I have a Fronius inverter that is not yet online so i'm just preparing things in HA for when i'm allowed to power it on.

When you say that overall usability is broken when you try to fix the error in this issue. What is it that gets broken? Isn't it enough to just skip trying to round/convert the state if there is no data to work with? Or if that is not enough set up "empty" data if no data could be fetched (with empty I mean indicating that all the sensors are zero).

I'm trying to see if I can fix these issues...but right now I have a problem where the "powerflow" data isn't polled at all...only the "inverterdata" is polled, as it should. Coding HA integrations is not my dayjob so i'm still struggling to understand their async core. But I cannot understand why only one of the data objects are polling :/

safepay, 2019-11-24:
> I have a problem where the "powerflow" data isn't polled at all...only the "inverterdata" is polled, as it should

First thing is to test whether the direct URL gets any data from the Powerflow:
http://{}/solar_api/v1/GetPowerFlowRealtimeData.fcgi

Just put your IP address in the right place and try in your browser. Eg:
http://192.168.0.1/solar_api/v1/GetPowerFlowRealtimeData.fcgi

As for nighttime errors, I 100% agree that it should just skip over if it can't connect.
I just updated the code as I think I had a wrong ConnectionError handling section which I've just tweaked.

nilrog, 2019-11-24:
I know it is not getting anything...the inverter is offline :)
Which is the same as what will happen during night, afaik. Unless you configure your inverter to be always online. I expect that HA will continue to poll following the throttle settings, but it only polls one of them.

Right now I reverted back to your code without any of my changes and then I see none of them polling...which looks weird. Both calls the inverter once, at setup, but after all those exceptions due to no state, there is no more polling.

Guess it's time to sleep and continue on this tomorrow :P

nilrog, 2019-11-25:
Ok, I think I got my head wrapped around this now and it looks ok when the inverter is offline. But I would like to try it when it's online to see if something is broken then also :P
But that won't happen until earliest on Wednesday this week if the power company does what they are supposed to.

Anyway, I created a fork and uploaded my changes there if you want to look at it and maybe try it.
https://github.com/nilrog/sensor.fronius

There were two issues...
- The exception handling around the requests was not ok. Even with your latest change it threw lots of errors.
- In the sensor class the local variable 'state' could be undefined before it was used.

With these changes there is not much in the logs when the inverter is offline. It only logs that it was unable to fetch the data, everytime it tries to contact the inverter. And it tries to fetch from both urls now when all exceptions are gone.
The power-wheel-card is also happy because the sensors it needs exists...with state 'unknown' so the card shows 'NaN' in the wheel.

But, as I said, I do NOT know how it behaves if the inverter is online...yet. So I might have broken something :| Which is why I haven't sent you a pull-request yet.

safepay, 2019-11-26:
Looks good to me and still retrieves data. I think that's where I was going with the error handling, but the state changes look good.

I'll upload it to the main branch as I've tested it and it seems Ok.

No need to submit a pull request at this point.

nilrog, 2019-11-26:
Good, thanks for the testing :)
I usually don't like to push things that I haven't been able to test myself.

Looking forward to tomorrow when I am hopefully allowed to turn on my inverter...and use your sensor for real.

jaganath-b, 2019-11-26:
Great, thanks a lot. No more recurring errors in the log :-)

safepay, 2019-11-27:
Changes have been merged in to current master version. Thanks all for helping.

closed 2019-11-27 by safepay