safepay/sensor.fronius #12: scan_interval in sensor.yaml not working

stavroz, 2019-12-02:
If I change the scan_interval in HA sensor.yaml config to 10 seconds it wont work.

if i go to the sensor.py in custom_components\fronius_inverter and change to the following:
MIN_TIME_BETWEEN_UPDATES = timedelta(seconds=10)

It works and changes the interval to 10 seconds.

It does end up eventually changing back however to 60 seconds, i think on next restart.

nilrog, 2019-12-02:
The `scan_interval` can never set the interval to anything less than what is defined in the integration, which is currently defined as:
`MIN_TIME_BETWEEN_UPDATES = timedelta(seconds=60)`

I don't like this hardcoded limit either and would rather see that it is removed, or set to the lowest setting that is feasible (maybe 10 seconds), because I too want it lower than 60 secs, And that the documentation is updated to highlight that you can control it with the `scan_interval` setting. And also what drawbacks it can have if you set it too low.

The other part, where you say that it goes back to 60, I have never seen. Are you sure that didn't happen when you updated the integration to a new version?

stavroz, 2019-12-04:
You might be right about that. I will keep an eye on it if the integration is updated.
thanks.

closed 2019-12-04 by stavroz