safepay/sensor.fronius #13: Platform error: sensor - Integration 'fronius_inverter' not found.

pesor, 2019-12-02:
I am running version Home Assistant 0.102.2, on a Synology Docker, and I keep getting this error, when trying to install this components.

Platform error: sensor - Integration 'fronius_inverter' not found.
1. december 2019 14.25 config.py (ERROR)

I even tried it on a Ubuntu 18.4, and get the same error.

I have followed the Read.me, and everything seems to be set up correctly.

What am I doing wrong?

Regards,
Per

nilrog, 2019-12-02:
Did you restart HA after copying this integration into custom_components, before you try to configure it?

I’m running the same setup and it works without that error.

pesor, 2019-12-03:
Just removed everything from my configuration.yaml, except the initial configuration.
Restarted home assistant,
added the fronius_inverter in the config/custom_components folder
Restarted home assistant
added:

sensor:
  - platform: fronius_inverter
    ip_address: 192.168.1.239

in the configuration.yaml folder

Restarted home assistant

get this error:

Platform error: sensor - Integration 'fronius_inverter' not found.
10.20 config.py (ERROR)

it is driving my crazy, as it should work, but it does not.

Any ideas?

nilrog, 2019-12-03:
What is in the `config/custom_components/fronius_inverter/` ? Could you post the output of `ls -l`?
It sounds like something is not in the correct place.

pesor, 2019-12-03:
Here are the outputs from Terminal:
bash-5.0# ls config                                                             
custom_components                                                               
bash-5.0# cd custom_components                                                  
bash: cd: custom_components: No such file or directory                          
bash-5.0# ls                                                                    
@eaDir                documentation         scripts.yaml                        
automations.yaml      groups.yaml           secrets.yaml                        
config                home-assistant.log    tts                                 
configuration.yaml    home-assistant_v2.db                                      
deps                  scenes.yaml                                               
bash-5.0# ls -l config                                                          
total 0                                                                         
drwxr-xr-x    1 1027     users           32 Dec  3 10:07 custom_components      
bash-5.0# cd config                                                             
bash-5.0# ls                                                                    
custom_components                                                               
bash-5.0# cd custom_components                                                  
bash-5.0# ls                                                                    
fronius_inverter                                                                
bash-5.0# cd fronius_inverter                                                   
bash-5.0# ls                                                                    
@eaDir         __init__.py    manifest.json  sensor.py                          
bash-5.0# ^C

pesor, 2019-12-03:
sorry here is the ls -l

bash-5.0# ls -l                                                                 
total 20                                                                        
drwxrwxrwx    1 root     root           144 Dec  3 10:07 @eaDir                 
-rwxr-xr-x    1 1027     users           29 Dec  1 14:24 __init__.py            
-rwxr-xr-x    1 1027     users          224 Dec  1 14:24 manifest.json          
-rwxr-xr-x    1 1027     users        10595 Dec  1 14:24 sensor.py

nilrog, 2019-12-03:
The file `init.py` should be named `__init__.py`
Try renaming it, restart HA and see if it starts working.

pesor, 2019-12-03:
I do not know how i listed as init.py, as it have been __init___.py the whole time.

Now I did a total restart removing all, and adding all, as per your first response.

Now the files lists:
bash-5.0# cd config                                                             
bash-5.0# cd custom_components                                                  
bash-5.0# cd fronius_inverter                                                   
bash-5.0# ls -l                                                                 
total 20                                                                        
drwxrwxrwx    1 root     root           144 Dec  3 11:01 @eaDir                 
-rwxrwxrwx    1 1027     users           29 Dec  1 14:24 __init__.py            
-rwxrwxrwx    1 1027     users          224 Dec  1 14:24 manifest.json          
-rwxrwxrwx    1 1027     users        10595 Dec  1 14:24 sensor.py              
bash-5.0#                                                                       
 
and it still do not find it:
2019-12-03 11:07:10 ERROR (MainThread) [homeassistant.config] Platform error: sensor - Integration 'fronius_inverter' not found.

I also changed to 777, in order not to have a security issue.

pesor, 2019-12-03:
Arh, the __ disappear in my copy / paste, it is in the ls -l 

[image]

pesor, 2019-12-03:
Right now, I will start from scratch, deleting the container, and build a new Home Assistant, from scratch.

I return with my results.

nilrog, 2019-12-03:
Ok, I guess GitHub's editor tries to parse it...you could try the `code` formatting when you post command output.

Anyway, a longshot...try deleting the @eaDir folder that one of the Synology packages put there. It should not make a difference, but I don't have them anywhere in my "docker" folder on my Synology.
If that doesn't help could you post part of the `home-assistant.log` where it throws the error? Maybe there is something useful there.

pesor, 2019-12-03:
This is driving me crazy.

I created a complete new container, used a different folder name than previously, started home assistant, moved the fronius_inverter to config/custom_componens, restarted, edited the configuration.yaml file with the platform and ip address.

SAME ERROR

This is the content of the home-assistent.log:
2019-12-03 10:44:33 ERROR (MainThread) [homeassistant.config] Platform error: sensor - Integration 'fronius_inverter' not found.

on the logs page it states:
Platform error: sensor - Integration 'fronius_inverter' not found.
11.44 config.py (ERROR)

[image]

I have an idea, I would like to try to put in another custom_component, can you give me a name for it, and then I can see if it is recognised?

nilrog, 2019-12-03:
Hmm, weird...I only have one more custom component loaded and that is one I created myself.

You could try one from here:
https://github.com/custom-components

Or this one that I tried, but could not use because it lacked features (but it worked):
https://community.home-assistant.io/t/minecraft-server-sensor/320

pesor, 2019-12-03:
Hi, I solved it.
Problem is the following.

My full path to custom_components on the synology is homeassistant/config/custom_components,

Here is the misunderstanding, I moved custom_components to homeassistant/custom_components, and it works.

So the confusion is about the setup of the /config in the docker, as that is very confusing, also I have been wondering why the configuration.yaml was placed in homeassistant/ and not homeassistant/config, this was what got me thinking.

Thanks VERY much for your help, knowing that it is working in same setup, gives only one possibility, and that is that the error is on my site, which it was.

I am so happy.

nilrog, 2019-12-03:
Yay, welcome to the club running this Fronius sensor :)

Yes, docker setup can be confusing at first...until you understand the logic. The name of the host folder may, or may not, be the same name as the folder inside the container. I have the same setup as you, the local `/<path-to>/home-assistant/` is `/config` inside the container.

pesor, 2019-12-03:
Thank, just one question more.
Where do I change so that the result is in kWh, for the produced year and total.

I did try to change here:
CONF_NAME = 'name'
CONF_IP_ADDRESS = 'ip_address'
CONF_DEVICE_ID = 'device_id'
CONF_SCOPE = 'scope'
CONF_UNITS = 'units'
CONF_POWER_UNITS = 'power_units'
CONF_POWERFLOW = 'powerflow'

but it gives no result.

nilrog, 2019-12-03:
Use `units: 'kWh'` in your configuration.

But there are, imo, some issues with the energy sensors. The 'day' energy is not converted to kWh. And if you look at the details of the sensors they may appear to be using 'W' as unit. I created a [pull request](https://github.com/safepay/sensor.fronius/pull/11) for changing it and I am awaiting feedback on it.

nilrog, 2019-12-03:
Btw, which inverter do you have, and do you have the SmartMeter also?

safepay, 2019-12-05:
> Use `units: 'kWh'` in your configuration.
>
> But there are, imo, some issues with the energy sensors. The 'day' energy is not converted to kWh. And if you look at the details of the sensors they may [...]

This is now merged with the master branch. Sorry, but I've have been unavailable for a while.

safepay, 2019-12-12:
Closing as this is more a platform issue than a bug.

closed 2019-12-12 by safepay