safepay/sensor.fronius #14: Getting unnecessary errors while the inverter is offline by nights

omitoma, 2019-12-13:
While the inverter is automatically offline (by nights and by long sunless days of nordic winter when the inverter can't produce enough power to turn on) sensor can't connect. That is just a fact, not an error. Allowing to pause polling while sun is under horizon could be useful. But I don't know how we could recognise those _almost_ sunless days. That would be useful too. But trusting "could data" might be an issue.

nilrog, 2019-12-13:
With the latest changes to this sensor it should not log so much errors when the inverter is offline and it will report valid values (zero) if HA is restarted while the inverter is offline. I submitted some updates to this sensor to mitigate those errors since I too live in the nordic region where we now have little to no sun.

But not being able to contact the inverter is an error from HA's perspective even if we know that it is not an error. So not logging connection errors would make troubleshooting harder when you are trying to set things up.

Maybe it could be changed so that the connection errors can be supressed (not written to the log, possibly controlled by a config setting). And a new sensor could be added along all the other  sensors that publishes the state of the inverter? Then you could still see if the sensor is able to connect or not without having to log it all the time. Something like "inverter_online" that could be "True" or "False"?

That said, you could also have the logger in the inverter to be always online even if the actual power inverter is off due to lack of solar power. There is a setting for that in the inverter. I have it enabled because I have a Smart Meter installed that is also logging power consumption 24/7 and I am pulling in that data into HA 24/7. The drawback there is of course that the inverter will consume a small amount of power to keep the logging function alive.

omitoma, 2019-12-13:
Yes, logging is better than it was originally. Thank You for that. :-) 

But If we know the sun is under horizon, why should we poll inverter?

I made my own scripts and automations to extend polling interval while the sun is under horizon. But scripting, it is not a trivial task for everyone. And I don't know if extending polling interval is the best way to accomplish this. I assume it's not. But it was the only option I could to come up with and implement (through trial and error). It's ugly and heavy, so don't ask me to share it. ;-)

nilrog, 2019-12-13:
Some of us (me at least) want this sensor to pull 24/7...since I am fetching data from the Smart Meter connected to the inverter :)

But it would be nice if we could figure out a way so that all of us could have it setup the way we want to. I'm willing to help out with implementation if we can agree on how we should do it.

omitoma, 2019-12-13:
Well, I'm using [Sun integration ](https://www.home-assistant.io/integrations/sun/ )to set `scan_interval` to 99999 when `sunset` triggers and back to 60, when `sunrise` triggers,  Hassio checks sun's position when it (re)starts using `sun elevation below/above horizon` and set's `scan_interval` accordingly. That is how I have done with my what not. Should we find a better "how"?

And with my proposal in the sensor this behaviour should be an option with warnings:
- You MUST activate sun integration
- You MUST set your home's correct GPS location

safepay, 2019-12-16:
This sensor changes throttling to avoid errors:
https://github.com/home-assistant/home-assistant/blob/9d583ad9f98cd3f1c2c6832206c039712c0fe4db/homeassistant/components/awair/sensor.py

I'm happy if someone wants to try and implement something like this.

dkalinai, 2020-01-06:
Still an issue for me... :( 

Mon Jan 06 2020 21:59:36 GMT+0200 (Eastern European Standard Time)
Unable to connect to Fronius: HTTPConnectionPool(host='192.168.2.xxx', port=80): Max retries exceeded with url: /solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=CommonInverterData (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f81333990>: Failed to establish a new connection: [Errno 113] Host is unreachable'))

safepay, 2020-01-06:
Please note that there is no intention to resolve this issue at this time.

When an API endpoint goes offline for hours at a time, HA is not really able tto handle it.

If anyone has the skills to fix this and make a pull request, please go ahead.

Deez73, 2020-05-04:
Guess this issue is also resolved with #29, so this issue should be closed.