safepay/sensor.fronius #15: get rel_Autonomy and rel_SelfConsumption values from powerflow

pcordonet, 2020-01-01:
Hi, 
I make a powerflow query i have this result:
`{
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 78,
               "E_Day" : 2104.699951171875,
               "E_Total" : 2001023.125,
               "E_Year" : 2104.699951171875,
               "P" : 0
            }
         },
         "Site" : {
            "E_Day" : 2104.699951171875,
            "E_Total" : 2001023.125,
            "E_Year" : 2104.699951171875,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : 430.77999999999997,
            "P_Load" : -430.77999999999997,
            "P_PV" : null,
            "rel_Autonomy" : 0,
            "rel_SelfConsumption" : null
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2020-01-01T21:04:20+01:00"
   }
}`

Can you add this two variables?

This is a description of these variables:
# mandatory field
# available since Fronius Hybrid version 1.3.1-1
# available since Fronius Non Hybrid version 3.7.1-2
# current relative self consumption in %, null if no smart meter is connected
number rel_SelfConsumption;

# mandatory field
# available since Fronius Hybrid version 1.3.1-1
# available since Fronius Non Hybrid version 3.7.1-2
# current relative autonomy in %, null if no smart meter is connected
number rel_Autonomy;

Thank you.
Peter.

safepay, 2020-01-01:
Thanks Peter.

Those two items are in the API spec but have been left out deliberately because they don't seems to add any value to displays or automation.

Can you please justify why they should be added?

pcordonet, 2020-01-02:
Hi ,
I have a smartmeter.
One of two values, now i don't remember, define of percentatge of generation is possible.
I have a air condicionate, if i have this value i can calculate if is possible send on or send off, i calculate necessari 1000w to start, i want to know i will have.
Sorry for my english.
Peter

safepay, 2020-01-02:
No worries.
I've added those two to the master branch.
If this is working, please report back.

pcordonet, 2020-01-02:
Thank you.
I report the results.

pcordonet, 2020-01-27:
Hi, 
There isn't new release. Is possible the code not updated? 
I call update from HACS.
Thank you.

safepay, 2020-03-13:
No. It's in the master waiting for feedback before I create a new release.

nilrog, 2020-03-23:
I just synced my fork against your repo and I noticed that in this commit you also added some code for sunrise/sunset. But it has nothing to do with these two values from the Fronius API, nor is it documented or I fully understand if the code is complete or not.
Would you mind explaining it? :)

safepay, 2020-03-29:
Was trying to work out how to stop logging overnight.
I think the solution is to change the async methods to incorporate the device unavailable functions.

nilrog, 2020-03-29:
But that will mess with those that want this sensor to be available 24/7? Unless there is a config option to turn that feature off?

What about only logging those errors at debug level?

safepay, 2020-03-29:
The errors are logged by HA, not the integration. As soon as a data source is unavailable, HA will log connectivity errors.

I won't change this part of the code because I don't have the time.

colwilliamsnz, 2021-09-05:
Close. Entities added in previous release.

closed 2021-09-05 by colwilliamsnz