safepay/sensor.fronius #17: Total and year energy issue

planet4, 2020-03-23:
First of all thanks for sharing this! I have some issues with the total and yearly sensor. The first image is the integration instalkled via HACS. The second my old one where I use the API directly. I have tried to change the units but it does not report correct.

My total production is 10.39 MWh when I look in Solar webb.

[image]
[image]

planet4, 2020-03-23:
Maybe I am just stupid. I think it has been Wh all the time but my sensor diveded it by 100. I think you can trash this issue.

nilrog, 2020-03-23:
Nope, you are right in that the latest released version doesn't convert energy correctly. The current master has fixes for it but there has not been any new release for a while.

@safepay I think you should release the current code so those installing this through HACS will get all the latest bugfixes/features.

sparkydave1981, 2020-03-26:
Daily energy really needs to be in kWh and yearly / total in MWh

safepay, 2020-03-29:
Agree sparkydave.

I've updated the default energy to kWh for daily, but left yearly and total at MWh.

sparkydave1981, 2020-03-30:
Hi,
I updated the component and now it displays kWh rather than MWh for daily production which is great, however I think it's still calculating wrong as the value is 1000 times what it should be. ie: showing 1000kWh instead of 1kWh.

safepay, 2020-03-30:
Yes. I'm afraid that the conversion code no longer works as original.

Unfortunately I no longer have time to fix these issues as its got much more complex than the original.

Not sure whereto go from here, but ideally someone else that has beter debugging tools than me can take over. I just  use a text edior.

Apologies, bu too many cooks.....

nilrog, 2020-03-30:
Sorry for being the one who originally change the behavior of day energy.

I submitted a pull-request that restores the old check for conversion of day energy back to kWh, and not just the unit.
https://github.com/safepay/sensor.fronius/pull/20

I have not tested it yet. But you can either grab the sensor.py from my fork and try it if you have the possibility. Otherwise I will try to test it tonight and make sure it works and report back here.

https://github.com/nilrog/sensor.fronius/blob/master/custom_components/fronius_inverter/sensor.py

sparkydave1981, 2020-03-31:
last update works perfectly.  Thank you!

closed 2020-04-10 by safepay