safepay/sensor.fronius #21: REQUEST: daily_grid_usage and daily_house_load

Tovrin, 2020-04-03:
This is more of a request.  Not sure if this is possible or not, but could we get daily_grid_usage and daily_house_load so easy comparison could be made with daily_energy?

nilrog, 2020-04-03:
None of the Fronius API endpoints provides that data directly.

If you have a smartmeter you can get the accumulated energy for sold/consumed energy from grid. So you can use utility meter sensors in HA to calculate the daily energy sold/consumed with those as input. And with the daily sold/consumed sensors and the produced daily energy sensor you can calculate daily house energy usage with a template sensor.

I am currently not using exactly that setup in my solarpanel-package for HA. But you can look at it and get some inspiration for what you can do. I am soon going to update that file with the setup I described above.
[https://gist.github.com/nilrog/86e9ac714052e993ab5df78bcd782573](https://gist.github.com/nilrog/86e9ac714052e993ab5df78bcd782573)

That said, there could be some issues with the currently released version of this integration and utility meter sensors. It may sometimes return 0 for a sensor and if that sensor is used as input to a utility meter it will calculate wrong. But I am currently testing some changes to this integration that solves those problems. My only worry is if it breaks something for those who allows the inverter to shutdown during night. I will try to push those changes to my fork this weekend so other can grab it and test before I ask for it to be merged into this one.
We had a discussion around this in the HA forum a couple of days ago:
[https://community.home-assistant.io/t/fronius-inverter-integration/19274/243](https://community.home-assistant.io/t/fronius-inverter-integration/19274/243)

Tovrin, 2020-04-04:
Thanks for the prompt reply, nilrog.

Your integration is very reliable in that respect to graphing the period `grid_usage` and `house_load`  For daily reporting, I'm using kalkih's "mini_graph_card" plugin with an aggregate function of `max` which allows me to do a pretty accurate bar graph of `daily_energy`.  I'm trying to do the same with a daily `grid_usage` and `house_load`

I'm a bit of novice when it comes to HA and some of the discussions you linked to went a bit over my head.  I'm not quite to the programming level yet ( ... but I'll get there eventually).

I'll be interested to see what you come up with.

Beer money headed your way. :)

Michael

nilrog, 2020-04-04:
To be correct it is @safepay who deserves the credit for this integration and beer. I am just trying to help out.

Yes, doing stuff in HA can be a bit overwhelming. I remember that from when I started.
I will try to document my package file a bit more when I update it so it shows what all sensors are used for.

closed 2021-09-05 by colwilliamsnz