safepay/sensor.fronius #23: Take some rest between poll elements

nagyrobi, 2020-04-16:
From time to time I get null values for various parameters. I suspect this must be because during each poll the inverter is stressed three times: inverter_data, powerflow_data and smartmeter_data are pulled immediately after each other, which may cause the small webserver embedded in the unit to lock up for a while.

I suggest to add a 5-second sleep between the consecutive data pulls somewhere around lines 107 and 115.

I'd do it but I've zero experience with proper python programming.

nilrog, 2020-04-16:
There is a problem in the code, that was amplified with a pull-request I made a few month ago that causes this, together with the fact that in some API endpoints the inverter returns 'null' as value when the inverter is not working.

I have made a correction for this in my fork, and asked a few to test it before I submit a pull-request for it. I have been running it for at least 2 weeks now and I do not see any '0' values for sensors where I previously did so.

If you want you can try my changes and see if these gaps goes away. Post back here and I can prepare a pull-request if it works for you too. My fork is up-to-date with this one so the only change is the one that should address this issue.
[https://github.com/nilrog/sensor.fronius](https://github.com/nilrog/sensor.fronius)

Just grab the sensor.py and replace the one you have and restart HA.

nagyrobi, 2020-04-16:
OK, I just replaced my sensor.py with yours. I'll see how it behaves and post back.

nagyrobi, 2020-04-17:
I think it's safe to prepare a pull-request because it works fine here.

nilrog, 2020-04-17:
Ok, will do so this weekend. Thanks for the feedback :)

closed 2020-04-25 by nagyrobi