safepay/sensor.fronius #24: powerflow and smartmeter clarification

nagyrobi, 2020-04-16:
What's the difference between SmartMeter and PowerFlow? 

I have a Fronius Symo 3.7-3-M with a datamanager card and a Smart Meter 63A. Reading through the documentation there's a note`Set to True if you have a PowerFlow meter (SmartMeter)` - thus I've enabled both `powerflow` and `smartmeter` parameters. If I disable one of them, some data will be missing. In my understanding based on the above, if I have a Smart Meter, i also have a PowerFlow, right?

Why are these separate?

nilrog, 2020-04-16:
You need a SmartMeter to be able to query the Inverter for PowerFlow data. That is the data that tells you how the energy flows in your solar installation...which is needed if you want to use the lovelace power-wheel card.

The `smartmeter` parameter enables a separate API that allows you to get detailed readings about energy, power, AC voltage/current etc. that flows through the SmartMeter.

So if you are not interested in the detailed readings from the SmartMeter but you want to have the PowerFlow readings you can disable the SmartMeter part...and there is one less query that will be sent towards the inverter.

Since I submitted the SmartMeter part I can see if I can improve the documentation so it becomes more clear what each part does.

nagyrobi, 2020-04-16:
> Since I submitted the SmartMeter part I can see if I can improve the documentation so it becomes more clear what each part does.

That would be good. Thank you for the explanation.

nilrog, 2020-04-20:
I submitted an update this weekend that included some tweaks to the README.

closed 2020-04-25 by nagyrobi