safepay/sensor.fronius #26: AC Power can’t be 0 At night since last update

AbsolutJohan, 2020-04-19:
The last update was working fine, no drop outs, until the evening. The panels are not producing anything but AC power can’t go down to 0. So the whole power wheel and other things shows wrong data. 
Otherwise, great work!!
[image]

nilrog, 2020-04-19:
This a drawback from the Fronius API. In some of the endpoints it omits some data (like PAC in this case) when the inverter is not active, instead of continuing to send that data with the true value of 0.
This is once case I didn't look at when I fixed the latest bug so I can try to see if fixing this is easy and doesn't break anything else.

There are however two options you can use, for now at least.

1. Use `fronius_panel_status` instead of `fronius_ac_power` as input to the power wheel, like the README suggests.
2. Use `scope: System` instead of Device. This enpoint will return PAC (0) when the inverter is down.

AbsolutJohan, 2020-04-20:
Okey, thank you. I will use the fronius_panel_status instead!

closed 2020-04-20 by AbsolutJohan