safepay/sensor.fronius #27: Help: power-wheel-card Energy view?

nagyrobi, 2020-04-28:
Using the default config sample presented on the first page I don't get the Energy view:
`type: 'custom:power-wheel-card'
title: Solar Power
production_is_positive: false
solar_power_entity: sensor.fronius_panel_status
grid_power_entity: sensor.fronius_grid_usage
home_energy_entity: sensor.fronius_house_load`

Although in [power-wheel-card documentation](https://github.com/gurbyz/power-wheel-card/tree/master#card-parameters) it states that specifying `home_energy_entity` should be enough to see the Energy view.

Any idea on how to configure power-wheel-card with this integration in order to see the Energy view too?

nilrog, 2020-04-29:
I have never looked at what is needed to configure the energy view in the power-wheel card so I checked it now. And you have missed some details...it is not enough to have just the `home_energy_entity` to enable the energy view.

I don't know if the example in the readme is/was supposed to enable the energy view. But afaik it cannot work with the provided example. It will only give you the power view.

According to the power-wheel documentation you need to have one of these combinations to enable energy view:

1. Where you have separate energy entities for grid consumption/production.
```
solar_energy_entity:
grid_energy_consumption_entity:
grid_energy_production_entity:
```
2. Where you have only one entity for grid energy you also need a house energy entity.
```
solar_energy_entity:
home_energy_entity:
grid_energy_entity:
```

The integration itself does not give you the sensors needed if you want energy readings presented on daily basis. Which, imo, is what makes most sense.
So you need to create some additional sensors that will give you grid energy on daily basis.

I published a [gist with my HA-package](https://gist.github.com/nilrog/86e9ac714052e993ab5df78bcd782573) a while back that I use for all solarpower related setup in HA together with this integration. It is not fully up to date but the part you would need to copy is.
```
### energy calculations
utility_meter:
  # calculate daily energy consumed from grid
  grid_consumed_energy_day:
    source: sensor.fronius_smartmeter_energy_ac_consumed
    cycle: daily
  # calculate daily energy sold to grid
  grid_sold_energy_day:
    source: sensor.fronius_smartmeter_energy_ac_sold
    cycle: daily
```
So if you create these two sensors in your setup (with the smartmeter sensors enabled in this integration) it will give you two additional sensors that you can use to configure the power-wheel in lovelace and you will have both power and energy view.
```
# power entities
solar_power_entity: sensor.fronius_panel_status
grid_power_entity: sensor.fronius_grid_usage
# energy entities
solar_energy_entity: sensor.fronius_day_energy
grid_energy_consumption_entity: sensor.grid_consumed_energy_day
grid_energy_production_entity: sensor.grid_sold_energy_day
```
[image]

[image]

(I will create a pull-request to add this to the documentation later)

nagyrobi, 2020-04-29:
Many thanks for the assistance. 

I tried this. And I get 

> Attribute "unit_of_measurement" for the entity "sensor.grid_sold_energy_day" not found in HA.
Units not equal for all sensors for the energy view.

I suppose that's because my `sensor.grid_sold_energy_day` is always 0, and `sensor.fronius_smartmeter_energy_ac_sold` is also always 0. Why?

nilrog, 2020-04-29:
That error happens because the three energy sensors have different units (Wh, kWh or MWh). The two utility sensors will inherit the units from the underlying sensors from the integration...and those are controlled by the `units` variable in the integration config.

The `fronius_day_energy` is always in kWh and cannot be changed. Looking at the code it may be that the energy sensors that are added by the `smartmeter` config should have their own unit, separate from the `units` which is for the inverter year and total energy, that may be very large. I have all my sensors set to kWh so for me it just works.

What you can do is to take the two `grid_*energy_day` and feed them into template sensors that converts them to kWh before sending them to the utility meter sensors. Then the units will match.

Why your `sensor.fronius_smartmeter_energy_ac_sold` is zero, I don't know. That value comes from the inverter and is just converted to right unit by the integration.

You can check the raw value by opening this URL in your browser, connected to the same network as the inverter.

> http://inverter-ip/solar_api/v1/GetMeterRealtimeData.cgi?Scope=System

nagyrobi, 2020-04-29:
Here's what I get:
```
{
   "Body" : {
      "Data" : {
         "0" : {
            "Current_AC_Phase_1" : 1.5069999999999999,
            "Current_AC_Phase_2" : 1.6559999999999999,
            "Current_AC_Phase_3" : 3.1880000000000002,
            "Details" : {
               "Manufacturer" : "Fronius",
               "Model" : "Smart Meter 63A",
               "Serial" : "18460377"
            },
            "Enable" : 1,
            "EnergyReactive_VArAC_Sum_Consumed" : 82090,
            "EnergyReactive_VArAC_Sum_Produced" : 15467140,
            "EnergyReal_WAC_Minus_Absolute" : 6166643,
            "EnergyReal_WAC_Plus_Absolute" : 0,
            "EnergyReal_WAC_Sum_Consumed" : 6166643,
            "EnergyReal_WAC_Sum_Produced" : 0,
            "Frequency_Phase_Average" : 50,
            "Meter_Location_Current" : 1,
            "PowerApparent_S_Phase_1" : 348.87049999999999,
            "PowerApparent_S_Phase_2" : 381.70799999999997,
            "PowerApparent_S_Phase_3" : 718.89400000000001,
            "PowerApparent_S_Sum" : 1289,
            "PowerFactor_Phase_1" : 0.88,
            "PowerFactor_Phase_2" : 0.78000000000000003,
            "PowerFactor_Phase_3" : 0.98999999999999999,
            "PowerFactor_Sum" : 0.94999999999999996,
            "PowerReactive_Q_Phase_1" : -141.93000000000001,
            "PowerReactive_Q_Phase_2" : -205.90000000000001,
            "PowerReactive_Q_Phase_3" : -21.390000000000001,
            "PowerReactive_Q_Sum" : -369.22000000000003,
            "PowerReal_P_Phase_1" : -274.26999999999998,
            "PowerReal_P_Phase_2" : -258.17000000000002,
            "PowerReal_P_Phase_3" : -702.72000000000003,
            "PowerReal_P_Sum" : -1235.1600000000001,
            "TimeStamp" : 1588189857,
            "Visible" : 1,
            "Voltage_AC_PhaseToPhase_12" : 400.10000000000002,
            "Voltage_AC_PhaseToPhase_23" : 394.89999999999998,
            "Voltage_AC_PhaseToPhase_31" : 395.80000000000001,
            "Voltage_AC_Phase_1" : 231.5,
            "Voltage_AC_Phase_2" : 230.5,
            "Voltage_AC_Phase_3" : 225.5
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DeviceClass" : "Meter",
         "Scope" : "System"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2020-04-29T21:50:58+02:00"
   }
}

```

nilrog, 2020-04-30:
The meter reports zero for `EnergyReal_WAC_Sum_Produced` so the corresponding `sensor sensor.fronius_smartmeter_energy_ac_sold` will be zero.

nagyrobi, 2020-04-30:
Any idea why can this be? Is there any setting in Fronius inverter I should enable?

nilrog, 2020-04-30:
Are you sure you have exported any energy to the grid?

If you are sure that you have exported to the grid there is only one setting in the inverter that you can change that has any effect on this. And that is the position of the smartmeter in your system, If it is connected to the feed-in-point or the consumption-path. It has to be set to the correct value otherwise the inverter will not be able to correctly calculate how the power/energy flows in your system.

That is described in the installation manual for the smartmeter:
https://www.fronius.com/~/downloads/Solar%20Energy/Quick%20Guides/SE_QG_How_To_Install_A_Fronius_Smart_Meter_EN.pdf

nagyrobi, 2020-04-30:
Yes of course. The energy company's own utility meter measures separately the consumed and the produced energy, and there are produced figures.

The position of my SmartMeter is in the consumption path. That's how it's electrically connected and that's how it's set up in the system.

[image]

nagyrobi, 2020-04-30:
Also on the inverter's own web interface I see:
[image]

nilrog, 2020-04-30:
Ok, then the reason why it's zero is that the smartmeter is in the consumption-path. With that setup it has no knowledge of how much energy that is exported to the grid since it can only measure the consumption. So you need different sensors to feed the power-wheel than I do since my meter is in the feed-in path.

You need to create your own sensor(s) that calculates the grid consumed/sold energy because your smartmeter can only measure how much energy that your system consumes. These sensor(s) will be the difference between energy produced by the inverter and the energy consumed by your system. The result +/- will be what is consumed or sold towards the grid. You can either have one sensor that can go negative or create two, one for consumed and one for sold.

nagyrobi, 2020-04-30:
But the inverter shows the feed in power.
Moreover, the feed in power is also available in the integration sensor, so the data is available. 
Can't it be calculated from existing data?

nagyrobi, 2020-04-30:
What makes me confusing then is what's sensor.fronius_grid_usage in my case. 
Se smartmeter in the consumption-path shouldn't be aware of that either.

nilrog, 2020-04-30:
You need to differentiate between power and energy...and what SolarWeb needs :)

Fronius provides data for power usage that will satisfy the needs of the power view in the power-wheel, if you have a smartmeter, through the powerflow API. But it does not provide the same for the energy view in that API. Probably because their SolarWeb does not display energy in that way.

You could probably fix it with something like this. But it is totally untested since I do not have your setup.

```
# the unit depends on your config and needs to match the fronius_day_energy which is in kWh
# i'm assuming kWh here, if not, either reconfigure the integration or create another
# template sensor that converts it to kWh
utility_meter:
  # convert consumed energy to daily energy (this is what the house consumes)
  house_energy_day:
    source: sensor.fronius_smartmeter_energy_ac_consumed
    cycle: daily

sensor:
  - platform: template
    sensors:
      # calculate grid energy (negative will be from grid, positive to grid)
      grid_energy_day:
        friendly_name: 'Grid energy'
        unit_of_measurement: 'kWh'
        value_template: '{{ states("sensor.fronius_day_energy") | float - states("sensor.house_energy_day") | float }}'
```

That should allow you to configure the power-wheel energy view like this:

```
solar_energy_entity: sensor.fronius_day_energy
home_energy_entity: sensor.house_energy_day
grid_energy_entity: sensor.grid_energy_day
```

nagyrobi, 2020-04-30:
Wow. Thanks. This seems to do the trick. The values are not correct though, I get 17.09 kWh for fronius_day_energy which seems correct, but I get 3924.09 kWh for house_energy_day and -3907.0 kWh for grid_energy_day which are abnormally high. I think there's a miscalculation by 1000 somewhere.

nagyrobi, 2020-04-30:
I think it's the  `sensor.fronius_smartmeter_energy_ac_consumed` the culprit. That's a value constantly growing from 0 when the meter was actually installed. The others are daily values.

nagyrobi, 2020-04-30:
This is what I have:
```
    units: 'kWh'
    power_units: 'W'
```
I'll see tomorrow how the counters will reset.

nilrog, 2020-04-30:
With that config you should be seeing correct values for grid and house also. But it may be that you got some "bad" data in there which will/should go away tomorrow.

nagyrobi, 2020-04-30:
ok, I'll let you know. Thx

nagyrobi, 2020-05-01:
Hi! Checked today, and values seem more appropriate. However there still seems to be an issue with the direction somehow. Check out:

Power view is OK:
[image]

Energy view doesn't show directions:
[image]

Money view all zeros:
[image]

Maybe one of the energy values should be negative?

nilrog, 2020-05-01:
Yay, it's getting close now :)

You probably need to invert the grid_energy_day sensor so that it is negative when you are exporting and vice versa.

Try changing it to this (division with -1):
```
grid_energy_day:
  friendly_name: 'Grid energy'
  unit_of_measurement: 'kWh'
  value_template: '{{ (states("sensor.fronius_day_energy") | float - states("sensor.house_energy_day") | float) / -1 }}'
```

nagyrobi, 2020-05-01:
Yep, we're getting closer 👍 

Here's my Energy view now:
[image]

Shouldn't the arrows be colored aswell? What's the normal behavior?
On your screenshot I see values near the arrows too. And my Money view is still completely gray.

nilrog, 2020-05-01:
Apparently you will not get colored arrows and values with the sensors we prepared for you. I found this in the power-wheel docs :(
> Nb. You will lack arrow coloring and arrow values in the energy view and money view due to supplying too less information to calculate these.

Try adding these two template sensors in addition to the other one. That should replicate what I have with the meter in the feed-in path.

```
grid_consumed_energy_day:
  unit_of_measurement: 'kWh'
  value_template: >
    {% if states("sensor.grid_energy_day") | float > 0 -%}
      {{ states("sensor.grid_energy_day") | float }}
    {%- else -%}
      {{ 0 | float }}
    {%- endif %}
```

```
grid_sold_energy_day:
  unit_of_measurement: 'kWh'
  value_template: >
    {% if states("sensor.grid_energy_day") | float < 0 -%}
      {{ states("sensor.grid_energy_day") | float * -1 }}
    {%- else -%}
      {{ 0 | float }}
    {%- endif %}
```

And change the power-wheel energy config to this:
```
solar_energy_entity: sensor.fronius_day_energy
grid_energy_consumption_entity: sensor.grid_consumed_energy_day
grid_energy_production_entity: sensor.grid_sold_energy_day
```

nagyrobi, 2020-05-01:
Hahaaaaaa now it's working!

Power View:
[image]

Energy View:
[image]

Money View:
[image]

Thanks a trillion for your excellent help!

nilrog, 2020-05-01:
You're welcome! :)

I will update the readme later with information on how to configure the energy view in the power-wheel and what additional template sensors one may need to create.

closed 2021-09-05 by colwilliamsnz