safepay/sensor.fronius #28: Error after update to Ha .109.0

suds32, 2020-04-30:
Error
Detected I/O inside the event loop. This is causing stability issues. Please report issue to the custom component author for fronius_inverter doing I/O at custom_components/fronius_inverter/sensor.py, line 325: result = requests.get(self._build_url(), timeout=10).json()

robthr, 2020-04-30:
I have the same issue running the latest version of fronius from HACS.

nagyrobi, 2020-05-01:
Something must have been changed in Home Assistant as this warning appears for multiple components

`2020-05-01 20:50:41 WARNING (MainThread) [homeassistant.util.async_] Detected I/O inside the event loop. This is causing stability issues. Please report issue to the custom component author for fronius_inverter doing I/O at custom_components/fronius_inverter/sensor.py, line 389: result = requests.get(self._build_url(), timeout=10).json()
`

lrybak, 2020-05-01:
Have the same
Begining 0.109 Home Assistant is checking for integrations with code which may cause potential stability and performance issue. Each time it should be reported to integration author

2020-05-01 21:12:53 WARNING (MainThread) [homeassistant.util.async_] Detected I/O inside the event loop. This is causing stability issues. Please report issue to the custom component author for fronius_inverter doing I/O at custom_components/fronius_inverter/sensor.py, line 325: result = requests.get(self._build_url(), timeout=10).json()

nilrog, 2020-05-01:
> Something must have been changed in Home Assistant as this warning appears for multiple components

Yes, as @lrybak wrote HA started to pinpoint integrations that are not fully async in 0.109.
Everything still works but this integration should be updated to fix this issue to avoid causing unnecessary strain on HA's event loop.

I may take a stab at doing that if I get some time over unless someone else does it.

nagyrobi, 2020-05-02:
Well, only this warning generates 1.3MB of log/day....

nilrog, 2020-05-03:
If someone is eager to test a version that has a fix for the above new warnings that HA started logging in 0.109 I have prepared an update here:
https://github.com/nilrog/sensor.fronius/tree/async-io

Just make sure that you grab the sensor.py version that is in the branch **async-io** that I linked to and not master and replace your local copy with that one.

Apart from making the requests async this version also contains some other changes compared to the current version available in HACS:

* All I/O requests towards the inverter are now async (no more warnings in HA).
* Sensor updates are now pushed to HA instead of pulled from HA.
* Support for `scan_interval` so you can configure how often you want to pull data from the inverter (default is 60s).
* Updated README with instructions for how to configure the energy view in the Power Wheel card in Lovelace.

I have been running it for the past day in a test setup with HA 0.109, since my main setup is still on 0.108.9, and it seems to work fine. But before I prepare a pull-request for this change I would appreciate if a few more people tried it out first and reported back here if all is ok or something is not ok.

Deez73, 2020-05-03:
I'll install the asyncio testversion right away, will get back in a day or so after the inverter gets back online when there's sunshine :)

lrybak, 2020-05-03:
Thanks for prompt reaction. I will test and report issues if any

lrybak, 2020-05-03:
Installed the asyncio version. Till now it works good. No I/O warning on event loop. Let's wait until sunshine for more valuable tests.

Btw. I found that sensors like PAC, UAC reports  as "unknown" in case the values are not returned by inverter API. Not sure if this is intended aproach.
@nilrog  wouldn't you consider returning in this case as 0. Unknown value is causing showing NaN value in power-wheel-card for example and I have to deal with extra template sensors to filter unknown values.
Thanks

nilrog, 2020-05-04:
@lrybak If you are using Power Wheel you should be using the powerflow sensors from the inverter. They are designed to support that use case and will always return a value, even when the inverter is down. See the readme for the supported way of configuring the Power Wheel.

The other APIs in the inverter may have parts, as you have discovered, that are omitted completely, or return "null", when that part of the internals in the inverter is not operational. Like when the sun is down the actual inverter circuitry is down.

Deez73, 2020-05-04:
Looking good so far, readings are coming in as expected, no errors so far (since last night after installing the asyncio version).

Good job @nilrog ... thanks mate!

nagyrobi, 2020-05-04:
Looking good here too. No more errors in the log, values feeding correctly the power wheel card and all my other usual graphs.
This is ready to be pushed in.
Thank you.

nilrog, 2020-05-04:
Thanks for the feedback :)
I have created a pull-request now.

lrybak, 2020-05-04:
Hello. Didn't noticed any issues with async version. Thank you

safepay, 2020-05-05:
I've merged this in to the master branch.
Let's wait a few days before a release.

closed 2020-05-05 by suds32