safepay/sensor.fronius #31: Error: "Detected I/O inside the event loop."

Maco65, 2020-05-15:
After recent upgrade of Home Assistant Core to  0.109.6 from 0.109.4 I have notices processor use increase from typical 1% to appr. 4% and the following logs
``` Logger: homeassistant.util.async_
Source: util/async_.py:120
First occurred: May 14, 2020, 4:23:32 PM (1846 occurrences)
Last logged: 8:16:30 AM

    Detected I/O inside the event loop. This is causing stability issues. Please report issue to the custom component author for fronius_inverter doing I/O at custom_components/fronius_inverter/sensor.py, line 325: result = requests.get(self._build_url(), timeout=10).json()
    Detected I/O inside the event loop. This is causing stability issues. Please report issue to the custom component author for fronius_inverter doing I/O at custom_components/fronius_inverter/sensor.py, line 356: result = requests.get(self._build_url(), timeout=10).json()
```
On the full log this error is reported 1-2 time every minute.
What can I do to fix it ?

nilrog, 2020-05-15:
A fix has been merged into master but it has not been released as an update yet (see #28). Until then you can manually copy sensor.py from master and restart HA and the I/O warning will go away.

Maco65, 2020-05-15:
Thanks. This link mentioned on the other issue: https://github.com/nilrog/sensor.fronius/tree/async-io does not work for me :( Where can I find fixed sensor.py file ?

nilrog, 2020-05-15:
Because it is in master of this repo now as I wrote.

But there is a new release out now so if you are using HACS just update it there :)

Maco65, 2020-05-15:
@nilrog  Thank you ! I am kind if new here so it seems I did not understand all correctly.
Anyhow updating to v0.9.3 (via HACS) has fixed the issue !
Excellent !

closed 2020-05-15 by Maco65