safepay/sensor.fronius #34: Problems losing entities

Tovrin, 2020-06-05:
I've been having problems with the integration losing the entities ... in that they become unavailable.  I can fix the problem by changing the name or increment a number at the end, but that loses all my history in the graphs.

It seems to happen on a complete reboot of the server or when patching Home Assistant.  It occasionally happens on a soft reboot.  It never used to be unstable, but I think it may be an issue with later patches breaking it.

nilrog, 2020-06-05:
I haven't seen that problem and I have been running with this code since I submitted the latest changes. I do not use the bleeding edge HA though. I am always running one major release after the current one because I do not have time to deal with all the issues that happens in every release. So I am currently using the latest 0.109.x version and will go to 0.110.x when 0.111.0 is released.

And I do restart HA at times since some changes require restarts. But I try to perform those updates "after dark" when the inverter is not producing...but it is logging 24/7 and all the sensors keep on working.

To see if we can figure out what you are experiencing we need more info.
Can you post this info and answer the questions:

- Your config for this integration
- Error logs, if they exists, from when this happens, or just the logs from when you have restarted HA and your sensors are unavailable.
- What version of HA you are using?
- Is your inverter logging 24/7 or only when it is producing?

Tovrin, 2020-06-05:
Config:
`- platform: fronius_inverter
  ip_address: <IP>
  name: <name>
  power_units: kW
  powerflow: true
  smartmeter: true
  monitored_conditions:
  - ac_power
  - ac_current
  - ac_voltage
  - day_energy
  - year_energy
  - total_energy
  - grid_usage
  - house_load
  - panel_status
  - rel_selfconsumption
  - smartmeter_energy_ac_consumed
  - smartmeter_energy_ac_sold`

Not sure how to get the logs.  I can't seem to find any.  IF you could point me in the right direction?

I'm currently running Home Assistant 0.110.4 on top of HassOS 4.9 and supervisor 227

I have a smart meter running, so the inverter is running 24/7.

I allocated some more memory to HA last night (up to 2G now) and it's running so much better whereas beforehand it was running like a dead dog.  I'm wondering if that may have been some of my issues.  I'm just trying a gratuitous reboot to see if that upsets it.  I'll let you know.

nilrog, 2020-06-08:
Ok, let's cross fingers hand hope that was it :)
I couldn't see anything strange in your config at least.

I have my HA running in docker on a Synology NAS. I don't remember when I changed, but it wasn't too long ago, the memory allocation to 2G. Before that I had 1.5G. Never really noticed that it has been memory constrained. But I have probably always kept it above what it has needed.

Deez73, 2020-07-21:
I have been experiencing a very similar issue, but running on a rpi3 ... after a HA restart (doing other HACS updates) the entities were not available and the sensor not working. What solved the issue was rebooting the host instead of just HA. Supervisor -> System -> Reboot (host system)