safepay/sensor.fronius #35: Using two Fronius Symo Inverters

Calvin4024, 2020-06-15:
Basically, I have two Symo Inverters with two different IPs
As long as I enter one IP or the other, everything is fine.... for one inverter at a time
I must do something wrong with the scope... should I enter both IP with a special syntax ? 
The Fronius software merge the data from the two inverters, I supposed that the scope: System would do the same or am I wrong ??
Thanks

nilrog, 2020-06-15:
As I have only one inverter I have never tried to setup this integration for two inverters so I cannot say for sure how it will behave or exactly how you should do it.. I also seem to remember reading in Fronius docs that with multiple inverters one is "master"? So that you query only one of the inverters but with different device_id for each of them.

If you use "scope: System" that should give you the totals for your system, for the limited number of sensors that are supported in that mode.

You should then also be able to configure two more integrations with "scope: Device" where one has "device_id: X" and the other "device_id: Y". You should also give them unique names with "name: inverter1" etc.

Try that and let us know how it worked out :)

Calvin4024, 2020-06-16:
Interresting, you put some ideas in my mind...

sensor:
  - platform: fronius_inverter
    ip_address: 192.168.1.60
    name: Fronius60
    powerflow: True
    power_units: kW
  - platform: fronius_inverter
    ip_address: 192.168.1.58
    name: Fronius58
    powerflow: True
    power_units: kW
  - platform: fronius_inverter
    ip_address: 192.168.1.60
    name: FroniusSystem
    powerflow: True
    power_units: kW
    scope: System

Now I have all the info I need, even more... basically, the entities under the scope: System have the same values as the name: Fronius60... to have global info, I will have to add the values of the two inverters, but that's fine....

Thanks for the idea...

[image]

nilrog, 2020-06-16:
You're welcome :)

I do think that your system should/could be configured so that the two inverters are connected together in Solar.web. Then you will have only one PoC that you can query for the three integrations, and the System one will give you the totals of the whole system.

Calvin4024, 2020-06-16:
I totally agree with you, and the two inverters are connected together in Solar.web... and I would like to have the right info without having to sum it up, but I don't know how to do it... as soon as I use scope: System, I only have the same value as one of the inverter... any better idea about a different syntax, maybe?

nilrog, 2020-06-16:
Ok, if both are connected together you should see the system summary for the ac_power, daily_energy, year_energy and total_energy sensors and not the same as for a single inverter. At least that is how Fronius describes the API and how this integration handles the response.

Can you post the output from this request?
http://<your-datalogger-ip>/solar_api/v1/GetInverterRealtimeData.cgi?Scope=System

Calvin4024, 2020-06-17:
I tried to access the API on both inverters.. and values are different...

http://192.168.1.xx/solar_api/v1/GetInverterRealtimeData.cgi?Scope=System

{
   "Body" : {
      "Data" : {
         "DAY_ENERGY" : {
            "Unit" : "Wh",
            "Values" : {
               "1" : 6436
            }
         },
         "PAC" : {
            "Unit" : "W",
            "Values" : {
               "1" : 120
            }
         },
         "TOTAL_ENERGY" : {
            "Unit" : "Wh",
            "Values" : {
               "1" : 3646781
            }
         },
         "YEAR_ENERGY" : {
            "Unit" : "Wh",
            "Values" : {
               "1" : 3646781
            }
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DeviceClass" : "Inverter",
         "Scope" : "System"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2020-06-17T07:42:30+02:00"
   }
}

And what Fronius software displays (2 secs later):

[image]

Calvin4024, 2020-06-17:
Basically, if I query both inverters and sum the PAC values, the result will correspond to the total displayed in fronius software

Calvin4024, 2020-06-17:
.. and about Year Energy, what Fronius says...

[image]

Calvin4024, 2020-06-17:
Just a thought... I have a Fronius Symo and a Fronius Symo Hybrid to charge my batteries....

nilrog, 2020-06-18:
Your two inverters are standalone devices that are "connected" together only by solar.web. So each one's System reading will report only that inverters totals. With this setup there is no need for the third integration (the System one). You can use the two Device integrations and sum it up in HA like you did.

Apparently to have them linked together so that the local API shows them both they need to be linked together and one will be master and the other slave. How to do that I do not know since I have only a single inverter.

Calvin4024, 2020-06-20:
Thanks for your help... you're absolutely right and I understand a bit better how it works now.... I'm using the API with a bit of math and I have all the info needed.... it is really an amazing tool... cheers

jurand71, 2021-07-03:
Hi,

In my case, when I use scope option set to System no other entities are recognized than: 

sensor.froniussystem_ac_current
sensor.froniussystem_ac_frequency
sensor.froniussystem_ac_voltage
sensor.froniussystem_dc_current
sensor.froniussystem_dc_voltage
But the state of them is unknown. When I change scope to Device everything is ok, but I receive data from the master inverter only.

My configuration is:

platform: fronius_inverter
ip_address: 192.168.1.xx
name: FroniusSystem
units: kWh
scope: System

I asked Fronius API by http://192.168.1.xx/solar_api/v1/GetInverterRealtimeData.cgi?Scope=System

The result is:

{
   "Body" : {
      "Data" : {
         "DAY_ENERGY" : {
            "Unit" : "Wh",
            "Values" : {
               "1" : 5814,
               "2" : 3689
            }
         },
         "PAC" : {
            "Unit" : "W",
            "Values" : {
               "1" : 3117,
               "2" : 1854
            }
         },
         "TOTAL_ENERGY" : {
            "Unit" : "Wh",
            "Values" : {
               "1" : 22360800,
               "2" : 106947
            }
         },
         "YEAR_ENERGY" : {
            "Unit" : "Wh",
            "Values" : {
               "1" : 2442399,
               "2" : 106947
            }
         }
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DeviceClass" : "Inverter",
         "Scope" : "System"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2021-07-03T10:01:11+02:00"
   }
}

What's wrong why I don't have daily, year and total entities? Could you help with this?

jurand71, 2021-07-06:
I change to standard fronius integration, and it works fine.

nagyrobi, 2021-09-05:
In theory you should connect the two inverters to each other via MODBUS (assigning them different IDs), and only connect one of them to the LAN network.

That way data for both of them will appear in the same API.