safepay/sensor.fronius #39: Adding MPPT1 and MPPT2 string volatge, watts, Temp - API call included

bobthebuilder2020, 2021-01-27:
Hi Safepay thanks for a great sensor in Home Assistant. Its working well for me, could you add this feature if possible?

Here is the code below for the API call.

// API URL - Fronius
$inverterDataURL = "http://".$dataManagerIP."/solar_api/v1/GetArchiveData.cgi?Scope=System&StartDate=".$date."&EndDate=".$date."&Channel=Voltage_DC_String_1&Channel=Current_DC_String_1&Channel=Voltage_DC_String_2&Channel=Current_DC_String_2&Channel=Temperature_Powerstage";

More info here, sorry im not a programmer so limited on explaining.

https://github.com/grann0s/_pushPVStringData.php/blob/master/_pushPVStringData.php

Bob

nilrog, 2021-06-02:
The problem here is that this is not part of the real-time API. This is archived data which, A: you cannot query as frequently, and B: you will also receive the data in batch format.

QbaF, 2021-06-02:
Ad. A. how frequently it is possible to query?

nilrog, 2021-06-02:
I think the smallest data chunks is 5 mins...if you query more frequently you will get the same data...but the problem is that this API will provide you with a set of readings, not a single reading for each parameter. So it is more intended for post-display/analyzis than showing data in HA.

Apparently this data can be obtained in realtime over modbus/tcp. There are some post over at Home Assistant that shows how they obtained some data that way.