safepay/sensor.fronius #44: fronius error

Cireasa23, 2021-04-09:
Logger: homeassistant.loader
Source: loader.py:786
First occurred: 9:40:27 AM (2 occurrences)
Last logged: 9:40:28 AM

No 'version' key in the manifest file for custom integration 'fronius_inverter'. As of Home Assistant 2021.6, this integration will no longer be loaded. Please report this to the maintainer of 'fronius_inverter'
No 'version' key in the manifest file for custom integration 'midea_ac'. As of Home Assistant 2021.6, this integration will no longer be loaded. Please report this to the maintainer of 'midea_ac'

algernon2607, 2021-04-10:
Hello
 I add      version key : "version": "v0.9.3"          in manifest.json and it works.
Be care to comma ...

Cireasa23, 2021-04-10:
Hey,
Thank you, the error is not shown any more.
Best regards,

closed 2021-04-10 by Cireasa23