safepay/sensor.fronius #47: No longer reporting power usage at night

stavroz, 2021-05-15:
For some reason after the update to fix the manifest I do not get power usage data when the panels are not powered overnight.

staticlama, 2021-05-15:
I'm having the same issue

rybackisback, 2021-05-15:
power, relative_autonomy, day_energy do not work in version 0.94. Downgraded back to 0.93

algernon2607, 2021-05-15:
In fact , it is the brbraga integration the cause of the break  in the night ...
You need to modify sensor.py manually ...

nilrog, 2021-05-15:
Exactly, this pull request that was also merged with the manifest update is the reason why it has stopped working in the night.
https://github.com/safepay/sensor.fronius/pull/45

There should be a configurable setting to control if the plugin should poll the inverter during night or not since there will be errors in the HA logs if the plugin is polling the inverter when it has shutdown for the night.

bamzero, 2021-05-17:
Is it a difference in firmware? I'm still on fro30381 on my inverter and 3.14.1-10 on my data manager card, perhaps the changes in #45 are only needed for newer firmware?

Anyway, I'll downgrade to 0.93 as well for now as not planning on upgrading Fronius firmware at the moment to find out.

nilrog, 2021-05-17:
No, there is no difference in firmware. The Fronius inverter, at least pre-gen24, as it is shipped from factory, shuts down the data manager when it stops producing electricity at night. But many, especially if you have a smartmeter, enables it to run 24/7. There is a setting in the inverter for that.

The code that was changed by #45 was the remains of an old attempt at trying to be smart about handling sunset/sunrise to reduce the error logs. But afaik it did not work out well so it was set to always return True and then no more effort was made to fix it. Until #45 showed up and it was merged along side the critical manifest update.

matake31, 2021-05-22:
Same problem here and I backed down to 0.9.3 as well.

Tovrin, 2021-05-22:
Ditto.  I've dropped back to 0.9.3 again.

thehaxxa, 2021-06-01:
I removed lines 232-234 from `sensor.py`:
```
if not self.available:           
    self._state = STATE_UNAVAILABLE
    return
```

Fixed the issue, now smart meter readings are recorded throughout the night.

closed 2021-09-04 by colwilliamsnz