safepay/sensor.fronius #52: Support Smart Meter TS 100A-1

blademckain, 2021-06-14:
Hi, really nice work
I have a Fronius inverter GEN 24 PLUS
with a Smart Meter TS 100A-1

i have added this config file to HA
```   
 - platform: fronius_inverter
    ip_address: myip
    powerflow: True
    power_units: kW
    smartmeter: True
    smartmeter_device_id: 0
```

and the result are:
[image]

as u see, many sensor, report UNKNOW status

this is the result from api
http://myip/solar_api/v1/GetMeterRealtimeData.cgi?Scope=Device&DeviceId=0
```   
{
   "Body" : {
      "Data" : {
         "ACBRIDGE_CURRENT_ACTIVE_MEAN_01_F32" : -5.7779999999999996,
         "ACBRIDGE_CURRENT_AC_SUM_NOW_F64" : -5.7779999999999996,
         "COMPONENTS_MODE_ENABLE_U16" : 1.0,
         "COMPONENTS_MODE_VISIBLE_U16" : 1.0,
         "COMPONENTS_TIME_STAMP_U64" : 1623686508.0,
         "Details" : {
            "Manufacturer" : "Fronius",
            "Model" : "Smart Meter TS 100A-1",
            "Serial" : "2651324894"
         },
         "GRID_FREQUENCY_MEAN_F32" : 50.0,
         "SMARTMETER_ENERGYACTIVE_ABSOLUT_MINUS_F64" : 355124.0,
         "SMARTMETER_ENERGYACTIVE_ABSOLUT_PLUS_F64" : 577337.0,
         "SMARTMETER_ENERGYACTIVE_CONSUMED_SUM_F64" : 577337.0,
         "SMARTMETER_ENERGYACTIVE_PRODUCED_SUM_F64" : 355124.0,
         "SMARTMETER_ENERGYREACTIVE_CONSUMED_SUM_F64" : 221.0,
         "SMARTMETER_ENERGYREACTIVE_PRODUCED_SUM_F64" : 576646.0,
         "SMARTMETER_FACTOR_POWER_01_F64" : -0.98599999999999999,
         "SMARTMETER_FACTOR_POWER_SUM_F64" : -0.98599999999999999,
         "SMARTMETER_POWERACTIVE_01_F64" : -1333.9000000000001,
         "SMARTMETER_POWERACTIVE_MEAN_01_F64" : -1425.8602006688966,
         "SMARTMETER_POWERACTIVE_MEAN_SUM_F64" : -1333.9000000000001,
         "SMARTMETER_POWERAPPARENT_01_F64" : 1352.0999999999999,
         "SMARTMETER_POWERAPPARENT_MEAN_01_F64" : 1441.9705685618733,
         "SMARTMETER_POWERAPPARENT_MEAN_SUM_F64" : 1352.0999999999999,
         "SMARTMETER_POWERREACTIVE_01_F64" : -221.19999999999999,
         "SMARTMETER_POWERREACTIVE_MEAN_SUM_F64" : -221.19999999999999,
         "SMARTMETER_VALUE_LOCATION_U16" : 0.0,
         "SMARTMETER_VOLTAGE_01_F64" : 241.80000000000001,
         "SMARTMETER_VOLTAGE_MEAN_01_F64" : 241.78963210702304
      }
   },
   "Head" : {
      "RequestArguments" : {
         "DeviceClass" : "Meter",
         "DeviceId" : "0",
         "Scope" : "Device"
      },
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2021-06-14T16:01:49+00:00"
   }
}
```

http://myip/solar_api/v1/GetPowerFlowRealtimeData.fcgi
```   
{
   "Body" : {
      "Data" : {
         "Inverters" : {
            "1" : {
               "DT" : 1,
               "P" : 1655
            }
         },
         "Site" : {
            "BatteryStandby" : false,
            "E_Day" : null,
            "E_Total" : null,
            "E_Year" : null,
            "Meter_Location" : "grid",
            "Mode" : "meter",
            "P_Akku" : null,
            "P_Grid" : -1054.5999999999999,
            "P_Load" : -600.47312011718759,
            "P_PV" : 1707.7088012695312,
            "rel_Autonomy" : 100.0,
            "rel_SelfConsumption" : 36.280760820686339
         },
         "Version" : "12"
      }
   },
   "Head" : {
      "RequestArguments" : {},
      "Status" : {
         "Code" : 0,
         "Reason" : "",
         "UserMessage" : ""
      },
      "Timestamp" : "2021-06-14T16:12:05+00:00"
   }
}
```

nilrog, 2021-06-15:
Unfortunately the GEN24 seems to have a different API compared to the previous SYMO inverters. The API document has been updated though...and contains references to GEN24. So they have done something. But I don't think they have shared the full description of the new API changes.

What I can see in the latest version of the V1 API doc is this:
- GetPowerFlowRealtimeData
  - The E_Day, E_Total and E_Year is always null on GEN24...this can be seen in your response.
- GetMeterRealtimeData
  - Fronius says "Examples for GEN24 are identical"...but based on your response it is NOT...the fields have completely different names.

beniwohli, 2021-07-07:
Supposedly, the July firmware update (1.13) is going to bring back compatibility with the "old" API. Why they bothered with a `v1` namespace in their API only to then break the API without bumping the version indicator is a question only the Fronius devs know...

closed 2021-09-05 by colwilliamsnz