safepay/sensor.fronius #53: Failed to update: connection error

Yagu-znal, 2021-07-05:
Thank you for this integration! Great stuff!

In my log I get this error:
`Failed to update: connection error`
and the relevant sensors stay unknown

When I go to the address mentioned in your readme I get a 404 error but both urls mentioned below it do. 
http://<froniusip>/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=CommonInverterData

404 Not Found

HomeAsst, 2021-07-16:
hi, 
I just installed this awesome integration but I have the same issue.

I installed it using HACS and the log says:
2021-07-16 15:05:51 ERROR (MainThread) [custom_components.fronius_inverter.sensor] Failed to update: connection error
2021-07-16 15:05:56 ERROR (MainThread) [custom_components.fronius_inverter.sensor] Failed to update: connection error

@Yagu-znal, the link in the gitHun page is not correct but it is correct in the sensor.py

just try: http://{YOUR_IP_ADDRESS}/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=0&DataCollection=CommonInverterData.

@safepay - Riccardo please can you help us out here. Grazie mille

Matt

HomeAsst, 2021-07-16:
Hi, 
I found the solution. in the config.yaml file you need to specify the ip address without http://
that was my mistake as well

- platform: fronius_inverter
    ip_address: 192.168.1.25

and you should be good to go

speedracer1190, 2021-08-01:
Thanks for this!   this was also my issue as well,  I assume the http:// parameter was added to the code since a lot of youtube walk throughs.

Yagu-znal, 2021-08-29:
I edited line 23 in /config/custom_components/fronius_inverter/sensor.py to 
`_INVERTERRT = 'http://<realfroniusip>/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DeviceId=1&DataCollection=CommonInverterData'`
I 'hardcoded' the fronius ip in the script itself. It works now.

closed 2021-09-04 by colwilliamsnz