safepay/sensor.fronius #55: Adding support for 2020.8 Energy dashboard?

Eonsim, 2021-08-05:
Has any one got time to add support for the new energy dashboard?
I think all that's needed is to add support for the new statistics framework  by adding to each sensor sensor a state_class that is set to measurement.

d3wy, 2021-08-05:
I'm working on it presently with no idea what I'm doing. It's (for me) not as straightforward as it seems but I'll keep trying.

Eonsim, 2021-08-05:
I'm also looking at the code, but not exactly clear how to introduce the state_class to the sensor.

d3wy, 2021-08-05:
I've tried using entity customization just to add state_class as per https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics to the day_energy sensor without any change/luck.

I've also tried adding this to the plugin:

`    @property
    def state_class(self):
        return measurement`

Without luck

Eonsim, 2021-08-05:
Apparently a last reset attribute needs to be assigned as well.

    attribute_templates:
      last_reset: '1970-01-01T00:00:00+00:00'
      state_class: 'measurement'

colwilliamsnz, 2021-08-05:
Awesome effort, thank you! As an interim workaround you could feed the power sensor values into an HA "integration" sensor and then into "utility" sensors...at which point the required attributes are set. Or use "customize.yaml" and add the attributes there.

Eonsim, 2021-08-05:
I have a rough work around using utility sensors + template sensors, but it's not working reliably:

See below:

 - platform: template
   sensors:
     energy_used:
       value_template: '{{states(grid_consumed_energy_day)}}'
       unit_of_measurement: kWh
       device_class: energy
       attribute_templates:
         state_class: 'measurement'
         last_reset: '1970-01-01T00:00:00+00:00'
     energy_sold:
       value_template: '{{states(grid_sold_energy_day)}}'
       unit_of_measurement: kWh
       device_class: energy
       attribute_templates:
         state_class: 'measurement'
         last_reset: '1970-01-01T00:00:00+00:00'
     energy_gen:
       value_template: '{{states(pv_gen_energy_daily)}}'
       unit_of_measurement: kWh
       device_class: energy
       attribute_templates:
         state_class: 'measurement'
         last_reset: '1970-01-01T00:00:00+00:00'

utility_meter:
  # calculate daily energy consumed from grid (input must be in kWh)
  grid_consumed_energy_day:
    source: sensor.fronius_smartmeter_energy_ac_consumed
    cycle: daily
  # calculate daily energy sold to grid (input must be in kWh)
  grid_sold_energy_day:
    source: sensor.fronius_smartmeter_energy_ac_sold
    cycle: daily
  pv_gen_energy_daily:
    source: sensor.fronius_total_energy
    cycle: daily

d3wy, 2021-08-05:
I came across similar, only concern so far is that the last reset might cause problems... tomorrow?

colwilliamsnz, 2021-08-05:
I don't think so...I'm using "daily" energy meters...and as I've had them enabled for a few months now, I have 30 days of history in the new energy panel.

nagyrobi, 2021-08-05:
https://github.com/home-assistant/core/issues/54047#issuecomment-893305838

thehaxxa, 2021-08-08:
> I came across similar, only concern so far is that the last reset might cause problems... tomorrow?

I am experiencing issues relating to resets, I'm using the config provided by: @Eonsim 

[image]

The daily reset (with the sensor adding additional attributes) results in data from the previous day carrying forward. 

----------------------------------

This is the grid consumption output: 
[image]

I am curious how @colwilliamsnz over came this problem, could you post your configuration?

I am thinking a solution might be to set an automation to reset the `utility meter` at 11:59:59PM each day.

Eonsim, 2021-08-08:
@thehaxxa I ended up switching to the main integration as I never quite got this working as well as I wanted it to. It's a pity because this version has much more reasonable names than the main one.

If you want to use the main one the following config works:

```
 - platform: fronius
   resource: http://192.168.xx.xx/
   scan_interval: 10
   monitored_conditions:
   - sensor_type: inverter
   - sensor_type: meter
   - sensor_type: power_flow
```
Then in the energy settings page:
```
Grid consumption: Energy real consumed Fronius Meter 0 http://
Return to grid: Energy real produced Fronius Meter 0 http://
Solar production: Energy total Fronius Inverter 1 http://
```

Note if you live in a country that isn't covered by Solar Forecast (for example NZ), you need to select "Don't forecast" under solar production or you won't get the solar production graph.

colwilliamsnz, 2021-08-08:
@thehaxxa here you go, I've edited out some extraneous cruft to do with my EV charger etc so hope I got it right...hope this helps :) 

### Template Sensors:

# Convert house power consumption to a positive and round to 0 dp
# Fronius sensor reports as a negative and to 2 decimal places (overkill when reporting watts)
- platform: template
  sensors:
    power_total_consumption:
      friendly_name: Total Consumption
      unit_of_measurement: "W"
      value_template: '{{ (states("sensor.fronius_house_load") | float * -1) | round | int }}'

# The next two template sensors simplify Fronius single sensor value that is positive if consuming
# energy from the grid, and negative if the solar is supplying excess to the grid

# Grid usage
- platform: template
  sensors:
    power_grid_consumption:
      friendly_name: Grid Consumption
      unit_of_measurement: "W"
      value_template: >-
        {% if states('sensor.fronius_grid_usage') | float > 0 %}
          {{ states('sensor.fronius_grid_usage') | round | int }}
        {% else %}
          0
        {% endif %}

# Grid feed-in
- platform: template
  sensors:
    power_grid_feedin:
      friendly_name: Grid Feed-in
      unit_of_measurement: "W"
      value_template: >-
        {% if states('sensor.fronius_grid_usage') | float < 0 %}
          {{ (states('sensor.fronius_grid_usage') | float * -1) | round | int }}
        {% else %}
          0
        {% endif %}

### Integration Sensors:

# Convert house power consumption to energy consumption
- platform: integration
  source: sensor.power_total_consumption
  name: energy_total_consumption
  unit_prefix: k
  unit_time: h
  round: 2
  method: left

# Convert solar power production to energy production
- platform: integration
  source: sensor.fronius_panel_status
  name: energy_solar_production
  unit_prefix: k
  unit_time: h
  round: 2
  method: left

### Utility Sensors:

energy_house_consumption_daily:
  source: sensor.energy_total_consumption
  cycle: daily

energy_solar_production_daily:
  source: sensor.energy_solar_production
  cycle: daily

thehaxxa, 2021-08-10:
> @thehaxxa here you go, I've edited out some extraneous cruft to do with my EV charger etc so hope I got it right...hope this helps :)
 
 Thanks @colwilliamsnz, that was very well documented and informative. It got me most of the way 🙂
 To expand on it I thought I would include a few more details, specifially how I got Solar production working and cost integration.
 Also I did not use Total Consumption in my Energy Dashboard. 
 
 I should note SignalCO2 works well, hopefully in the future Solcast is added as a forecast provider for AUS/NZ users. 
 
 
 ## Template Sensors:
 ### Convert house power consumption to a positive and round to 0 dp as Fronius sensor reports as a negative and to 2 decimal places (overkill when reporting watts)
 ```
 - platform: template
   sensors:
     power_total_consumption:
       friendly_name: Total Consumption
       unit_of_measurement: "W"
       value_template: '{{ (states("sensor.fronius_house_load") | float * -1) | round | int }}'
 ```
 
 ### The next two template sensors simplify Fronius single sensor value that is positive if consuming energy from the grid, and negative if the solar is supplying excess to the grid
 ```
 - platform: template
   sensors:
     power_grid_consumption:
       friendly_name: Grid Consumption
       unit_of_measurement: "W"
       value_template: >-
         {% if states('sensor.fronius_grid_usage') | float > 0 %}
           {{ states('sensor.fronius_grid_usage') | round | int }}
         {% else %}
           0
         {% endif %}
 
 - platform: template
   sensors:
     power_grid_feedin:
       friendly_name: Grid Feed-in
       unit_of_measurement: "W"
       value_template: >-
         {% if states('sensor.fronius_grid_usage') | float < 0 %}
           {{ (states('sensor.fronius_grid_usage') | float * -1) | round | int }}
         {% else %}
           0
         {% endif %}
 ```
 
 ## Integration Sensors:
 ### Convert house power consumption to energy consumption
 ```
 - platform: integration
   source: sensor.power_total_consumption
   name: energy_total_consumption
   unit_prefix: k
   unit_time: h
   round: 2
   method: left
 ```
 
 ### Convert solar power production to energy production
 ```
 - platform: integration
   source: sensor.fronius_panel_status
   name: energy_solar_production
   unit_prefix: k
   unit_time: h
   round: 2
   method: left
 ```
 ### Convert grid consumption to energy production
 ```
  - platform: integration
    source: sensor.power_grid_consumption
    name: energy_grid_consumption
    unit_prefix: k
    unit_time: h
    round: 2
    method: left
 ```

 ### Convert grid feedin to energy production
 ```
  - platform: integration
    source: sensor.power_grid_feedin
    name: energy_grid_feedin
    unit_prefix: k
    unit_time: h
    round: 2
    method: left
 ```
 
 
 ## Utility Sensors:
 ```
  energy_solar_production_daily:
    name: "PV Generation (Daily)"
    source: sensor.energy_solar_production
    cycle: daily
  energy_grid_consumption_daily:
    name: "Grid Consumption (Daily)"
    source: sensor.energy_grid_consumption
    cycle: daily
  energy_grid_feedin_daily:
    name: "Grid Export (Daily)"
    source: sensor.energy_grid_feedin
    cycle: daily
	
  ## Not used in Energy Dashboard
  energy_house_consumption_daily:
    name: "Total Consumption (Daily)"
    source: sensor.energy_total_consumption
    cycle: daily
 ```
 
 ## Cost Input Number Integration (takes cost in cents i.e. 16.95 -> $0.1695)
 ```
 input_number:
  # Grid Consumption Energy Cost (AUD/kWh) 
  grid_consumption_energy_cost:
    name: 'Grid Consumption Energy Cost'
    mode: box
    min: 0
    max: 1000
    step: 0.001
    unit_of_measurement: "AUD (cents)/kWh"
    icon: mdi:currency-usd
  # Grid Generation Energy Cost (AUD/kWh) 
  grid_generation_energy_cost:
    name: 'Grid Generation Energy Cost'
    mode: box
    min: 0
    max: 1000
    step: 0.001
    unit_of_measurement: "AUD (cents)/kWh"
    icon: mdi:currency-usd
 ```
 ## Converts from cents to Dollars (limitation of input number precision)
 ```
  - platform: template
    sensors:
       energy_consumption_cost_aud:
        icon_template: mdi:currency-usd
        value_template: >
          {{(states('input_number.grid_consumption_energy_cost')|float / 100)|round(4, 'ceil')}}
        unit_of_measurement: 'AUD/kWh'
      energy_generation_cost_aud:
        icon_template: mdi:currency-usd
        value_template: >
          {{(states('input_number.grid_generation_energy_cost')|float / 100)|round(4, 'ceil')}}
        unit_of_measurement: 'AUD/kWh'
 ```
 
 Then the `energy_solar_production_daily`, `energy_grid_consumption_daily` and `energy_grid_feedin_daily`. Are added to the Energy Dashboard. The template cost sensors are used for cost, alternatively a static value can be entered.

fondberg, 2021-08-14:
I would like it to be in the actual integration rather than messing around with too much template configuration.
In my project we just merged a PR which makes it "automatic". It is quite simple so please have a look: https://github.com/fondberg/easee_hass/pull/131/files

sparkydave1981, 2021-08-19:
I've added all the required attributes to the sensor.fronius_day_energy entity but it still won't let me add it the the HA Energy system.  Hopefully there is a proper fix possible.
[image]

safepay, 2021-08-19:
@sparkydave1981 
You might be able to add the sensor to a custom template sensor with the addition of
`state_class: measurement`

smallsam, 2021-08-22:
I believe the sensor name needs to start with "energy_", which seems redundant given the device class but it's what I needed to do for an individual device energy meter.

colwilliamsnz, 2021-08-24:
> I believe the sensor name needs to start with "energy_", which seems redundant given the device class but it's what I needed to do for an individual device energy meter.

Hi, I can confirm it doesn't need this at all. If you "inject" the missing attributed using customise.yaml it works fine:

```
sensor.fronius_day_energy:
  friendly_name: Fronius Solar Production
  icon: mdi:solar-power
  last_reset: "1970-01-01T00:00:00+00:00"
  device_class: energy
  state_class: measurement
```

colwilliamsnz, 2021-08-24:
> I would like it to be in the actual integration rather than messing around with too much template configuration.
> In my project we just merged a PR which makes it "automatic". It is quite simple so please have a look: https://github.com/fondberg/easee_hass/pull/131/files

I have a working version that just needs to be tidied up a bit (I ran it side by side with the existing component and changed names a bit)...I'm not a contributor nor terribly up to speed with Github (but I'm here to learn) so I'm not sure how I go about proposing said changes? As you say, it was quite simple and grateful for you showing the way :)

nilrog, 2021-08-24:
FYI the way integrations was supposed to add support for the new energy dashboard in 2021.8 has already been changed for 2021.9. And the current way is already deprecated and will be removed soon (if I remember right it will go away in 2021.11). So any integration that was updated for 2021.8 will have to change again or it will stop working.

For details, they initially required a **last_reset** attribute for energy devices in order for them to be available as sensors for the energy dashboard. But to make it simpler for integrations the **last_reset** has been replaced with a new **state_class** type.

nilrog, 2021-08-24:
> I'm not a contributor nor terribly up to speed with Github (but I'm here to learn) so I'm not sure how I go about proposing said changes?

@colwilliamsnz It's not so difficult. In the simplest form you fork this repo, make your changes, commit your changes, and push them to github. And when you are done you create a pull-request from your repo to this one on github. When that is done the maintainer, and others, can look at your changes and merge them as-is or suggest changes.

Here is a link to the docs for how to work with forks. But there are more topics worth reading in the same place.
https://docs.github.com/en/github/collaborating-with-pull-requests/working-with-forks/about-forks

If you aim to do more collaborative work then you should start thinking about branches and keeping your fork in sync with the original (this repo). It will make life much easier in the long run.

Shponzo, 2021-09-01:
Hi, I just updated to v0.9.6, but I'm not able to add a energy sensor to the energy dashboard. Do I have to wait before the sensor becomes available?
I just have an inverter, so no smartmeter etc.

djm193, 2021-09-01:
I also did the update to v 0.9.6 and now I am able to get 3 sensors into the energy dashboard. 
With v 0.9.5 I did not geht any sensor.

However the sensor for consumption and grid do not appear whereas I get correct information when I add the sensor in Lovelace. So problem seems to be the integration of the other fronius sensors in the energy dashboard.

Any Ideas why my smartmeter integration does not work?
my configuration.yaml looks like this:
#Fronius Wechselrichter
sensor:
    - platform: fronius_inverter
      ip_address: 192.168.178.10
      powerflow: True
      units: kWh
      power_units: kW
      smartmeter: True
      smartmeter_device_id: 0
      monitored_conditions:
      - ac_power
      - day_energy
      - year_energy
      - total_energy
      - grid_usage
      - house_load
      - panel_status
      - rel_autonomy
      - rel_selfconsumption
      - 
[image]

colwilliamsnz, 2021-09-01:
> Hi, I just updated to v0.9.6, but I'm not able to add a energy sensor to the energy dashboard. Do I have to wait before the sensor becomes available?

It should be available to select in the Energy settings after a HA restart. Which entity are you trying to add? Check it under "Developer Tools" -- it should show the following additional entity attributes if the 0.9.6 is adding them correctly:

state_class: measurement
device_class: energy
last_reset: 1970-01-01T00:00:00+00:00

Note that a daily entity may produce some odd results in the energy dashboard with HA 2021.8 which will be fixed in 2021.9.

[image]

Shponzo, 2021-09-01:
> > Hi, I just updated to v0.9.6, but I'm not able to add a energy sensor to the energy dashboard. Do I have to wait before the sensor becomes available?
> 
> It should be available to select in the Energy settings after a HA restart. Which entity are you trying to add? Check it under "Developer Tools" -- it should show the following additional entity attributes if the 0.9.6 is adding them correctly:
> 
> state_class: measurement
> device_class: energy
> last_reset: 1970-01-01T00:00:00+00:00
> 
> Note that an entity that resets daily is likely to produce some odd results (negative generation at midnight) in the energy dashboard with HA 2021.8. This should be fixed in 2021.9.
> 
> [image]

Hi,

Your screenshot helped me! I had just the same settings, except for unit_of_measurement.
For me, it was still MWh. After changing this to kWh and restart, it worked in the energy dashboard.

Thank you!

colwilliamsnz, 2021-09-01:
> Your screenshot helped me! I had just the same settings, except for unit_of_measurement.
> For me, it was still MWh. After changing this to kWh and restart, it worked in the energy dashboard.

Great to hear. That's my default config so I missed that requirement. I'll update the readme to include this vital bit of info :)

xxharjinder, 2021-09-04:
I have the v0.9.6 installed on 2021.9.2. Should I be able to use the sensors from this integration with energy dashboard? I don't seem to have sensor.fronius_house_load show up under consumption which is the sensor that gives the current house load for me but in negative value.

colwilliamsnz, 2021-09-04:
> I have the v0.9.6 installed on 2021.9.2. Should I be able to use the sensors from this integration with energy dashboard? I don't seem to have sensor.fronius_house_load show up under consumption which is the sensor that gives the current house load for me but in negative value.

Install the latest release (it provides support for HA 2019.9.x). Additionally you can't user power (W) entities with the energy dashboard...you need to select an energy (kWh) entity.

xxharjinder, 2021-09-04:
> Install the latest release (it provides support for HA 2019.9.x). Additionally you can't user power (W) entities with the energy dashboard...you need to select an energy (kWh) entity.

do I have to convert the negative value to positive too?

colwilliamsnz, 2021-09-04:
> do I have to convert the negative value to positive too?

You should use the following entities in the dashboard:

Solar production: total_energy
Grid consumption: smartmeter_energy_ac_consumed (if you have a smart meter)
Grid feed-in: smartmeter_energy_ac_sold (if you have a smart meter).

> edit: I am also getting the below warning
> The following entities do not have the expected state class 'total_increasing'
> sensor.fronius_day_energy (measurement)

Sounds like you are running the previous release of this integration. I've asked the owner of this repo to cut a new release so you can upgrade via HACS. Until that is done you will need to manually install the latest release component by downloading and adding to HA (technically only sensor.py is needed).

xxharjinder, 2021-09-04:
> You should use the following entities in the dashboard:
>
> Solar production: total_energy
> Grid consumption: smartmeter_energy_ac_consumed (if you have a smart meter)
> Grid feed-in: [...]
I do have a smart meter and I have the entities sensor.fronius_smartmeter_energy_ac_consumed and sensor.fronius_smartmeter_energy_ac_sold. their values are 10,517 kwh 19,612 kwh. Those look like the accumulated values and not daily values

colwilliamsnz, 2021-09-04:
> I do have a smart meter and I have the entities sensor.fronius_smartmeter_energy_ac_consumed and sensor.fronius_smartmeter_energy_ac_sold. their values are 10,517 kwh 19,612 kwh. Those look like the accumulated values and not daily values

Cool. Well you can use them OR dailies if you wish...it doesn't actually matter as 2021.9.x works perfectly with entities with a state class of "total_increasing" which you'll get when you upgrade this component to 0.9.7 :)

colwilliamsnz, 2021-09-04:
@Safepay would it be possible to get some access to close/clean up some of these obsolete issues? I note people raise issues but rarely close so they hang around and become irrelevant. Just a thought :)

colwilliamsnz, 2021-09-04:
Tidy-up close. Support added in #57

closed 2021-09-04 by colwilliamsnz