safepay/sensor.fronius #59: Sugerencia

ffcadenas, 2021-08-29:
Thank you very much for the development but the truth is that it was not easy for me to configure the application.
My inverter is a Fronius Primo 5:

I would recommend putting a configuration example with the following with smartmeter, since the units come in MWh:
- platform: fronius_inverter
  ip_address: ([IP_Fronius_smartmeter]
  smartmeter: True 
  powerflow: True
  power_units: kW
  units: kWh

In "utility meters" the solar production sensor is missing, also the application did not recognize nee sensors if I did not modify them as "device_class": energy and its "units" in kWh, (attention: Kwh is not worth it for de W without capital letters)
utility_meter

#Energy (safepay)
  # calculate daily energy consumed from grid (input must be in kWh)
  grid_consumed_energy_day:
    source: sensor.fronius_smartmeter_energy_ac_consumed
    cycle: daily
    device_class: energy
    unit_of_measurement: kWh
  # calculate daily energy sold to grid (input must be in kWh)
  grid_sold_energy_day:
    source: sensor.fronius_smartmeter_energy_ac_sold
    cycle: daily
    device_class: energy
    unit_of_measurement: kWh
  #calculate daily energy produced from solar panels
  solar_produced_energy_day:
    source: sensor.fronius_total_energy
    cycle: daily  
    device_class: energy
    unit_of_measurement: kWh

colwilliamsnz, 2021-08-30:
The power units are up to you to decide what you want. For my dashboards, I find W works best for power and kWh is best for power (but there are no wrong answers here).

```
- platform: fronius_inverter
  ip_address: 192.168.1.71
  powerflow: true
  units: kWh
  power_units: W
  smartmeter: true
  smartmeter_device_id: 0
```

0.9.6 has support for the energy dashboard in HA 2021.8. You need to use the following sensors:

Solar production: total_energy
Grid consumption: smartmeter_energy_ac_consumed
Grid feed-in: smartmeter_energy_ac_sold

dietlman, 2021-09-04:
Thanks for your great work! I have integrated my Fronius and have no issues, just one thing I would like to know. Since the latest HA update 2021.9 they also offer to read data from Solar Battery. Do you think you could provide integration for the Fronius Solar Battery as well? That would be really great to read data stored in the battery and consumed from the battery.
Thanks

colwilliamsnz, 2021-09-04:
> Thanks for your great work! I have integrated my Fronius and have no issues, just one thing I would like to know. Since the latest HA update 2021.9 they also offer to read data from Solar Battery. [...]

Did you want to create a new issue with this request in it? That way it has better visibility VS being buried in this non-related issue :)

closed 2021-09-05 by colwilliamsnz