safepay/sensor.fronius #62: WARNING: 2021.9 breaks Energy dashboard derived from this component

colwilliamsnz, 2021-09-01:
The developer docs for 2021.9 suggested that a planned change to energy sensors would be graceful allowing until 2021.11 for developers to make appropriate changes. This is not the case so suggest holding off on 2021.9 until a new release can be tested.

colwilliamsnz, 2021-09-01:
Test branch with HA 2021.9 support here: https://github.com/colwilliamsnz/sensor.fronius

@nilrog appreciate your thoughts on this one (and also the breaking changes noted in the readme). Concerned that HA 2021.9 broke the component (despite suggestions there would be some months of compatibility) and so how best to convey this as visibly as possible.

nilrog, 2021-09-02:
This is a bug in HA that they are not following their own documentation. It will be impossible for third-party integrations to solve this since not everyone will update to 2021.9.0 immediately. And if we were to release a new version through HACS for 2021.9.0 then people, that may still be on 2021.8.x, will update and we will be broken for them.

colwilliamsnz, 2021-09-04:
Close. Pull request merged https://github.com/safepay/sensor.fronius/commit/82000435a5cb0d47eb77f9ca189102cbca1319d2

closed 2021-09-04 by colwilliamsnz