safepay/sensor.fronius #64: Few issues with 0.9.7

cathelest, 2021-09-04:
I just installed 0.9.7,

Just a few questions.

1.
It seems for 
Grid consumption: smartmeter_energy_ac_consumed
Grid feed-in: smartmeter_energy_ac_sold
which are required for the energy dashboard, you need to specify
smartmeter: True

Please add those to sensors to the list on the github, as i was stumped for a few mins when they were not showing up, and as
it wasnt in the list on github i presumed they were default entities, (Thankfully i do have a smart meter) but if you dont
doesnt that mean you cant get those sensors, may need to mention that smartmeter is required under energy dashboard section.

2. Once the entities are created, it seems HA core-2021.9.2 (Current)
Wants the sensors to have a state class of state_class: total_increasing for all 3 sensors, 
easy fix using the customize.yaml, or could do via the gui,
sensor.fronius_total_energy:
  state_class: total_increasing
  last_reset: "homeassistant.util.dt.utc_from_timestamp(0)"
sensor.fronius_smartmeter_energy_ac_consumed:
  state_class: total_increasing
  last_reset: "homeassistant.util.dt.utc_from_timestamp(0)"
sensor.fronius_smartmeter_energy_ac_sold:
  state_class: total_increasing
  last_reset: "homeassistant.util.dt.utc_from_timestamp(0)"  

3. The last_reset
When i pull up the entity's in Developer Tools > States > Search for sensor.fronius_smartmeter_energy_ac_consumed
Under attributes (for all three customized entities)

state_class: total_increasing
attribution: Fronius Inverter Data
unit_of_measurement: kWh
friendly_name: Fronius SmartMeter Energy AC Consumed
icon: mdi:solar-power
device_class: energy
last_reset: homeassistant.util.dt.utc_from_timestamp(0)

Should the last_reset not be a date, eg 2021-01-01T00:00:01
Is that "homeassistant.util.dt.utc_from_timestamp(0)" line correct or hasnt been brought in properly??

colwilliamsnz, 2021-09-04:
> It seems for
> Grid consumption: smartmeter_energy_ac_consumed
> Grid feed-in: smartmeter_energy_ac_sold
> which are required for the energy dashboard, you need to specify
> smartmeter: True

That's correct (and implied by nature of the entity names ;) Clarification in the readme is certainly worthwhile if its not 100% clear.

> Once the entities are created, it seems HA core-2021.9.2 (Current)
> Wants the sensors to have a state class of state_class: total_increasing for all 3 sensors,
> easy fix using the customize.yaml, or could do via the gui...

You should't need to use customize.yaml. 0.9.7 adds that state_class attribute already to be compatible with HA 2021.9.x and is certainly the case on my system (IE, I can't reproduce what you're suggesting here).

> Should the last_reset not be a date, eg 2021-01-01T00:00:01
> Is that "homeassistant.util.dt.utc_from_timestamp(0)" line correct or hasnt been brought in properly??

This one needs investigation which I'll do. You could set to the epoch but refer previous comment ;)

cathelest, 2021-09-05:
Sorry about that pull request, 
I think i mucked myself up when the smartmeter entitys didnt show up with adding the old customize code, 
Closing the request, But by all means please add the clarification for smart meter as that would have definently helped, 
Sorry and thanks again,

closed 2021-09-05 by cathelest