safepay/sensor.fronius #66: Fronius Battery Support in Energy Manager

dietlman, 2021-09-05:
Thanks for your great work! I have integrated my Fronius and have no issues, just one thing I would like to know. Since the latest HA update 2021.9 they also offer to read data from Solar Battery. Do you think you could provide integration for the Fronius Solar Battery as well? That would be really great to read data stored in the battery and consumed from the battery.
[image]

Thanks

nilrog, 2021-09-07:
Possible, yes. But I don't think anyone of us who contributes code has a Fronius inverter with battery. So implementation will be "blind" and has to be tested by someone else.

And looking at the API docs the only inverter that is currently supported is the Symo Hybrid. Because the Gen24 needs FW 1.13 according to the docs, and that version is not yet released by Fronius. That FW version is also supposed to fix some issues where Fronius broke their own API in the Gen24.

dietlman, 2021-09-07:
I understand, yes you are right you need to have a Fronius Symo Hybrid to be able to connect a Battery, this is actually what I have, and I would be willing to test of course.  I also read that the native Fronius integration already supports the Battery as well, but I have not tried yet.

Kilowatt-W, 2021-10-07:
In Austria there are many Users with Symo Hybrid and even with the BYD Battery Box HV.

Actual Firmware is 1.18.7-1 and the rest of the integration is working like a charm :-)

I would like to test too if you need support.

lochstar, 2021-11-21:
I own a Gen24 with Battery and can help test/develop.

Zelete, 2021-12-10:
Also Gen24 with BYD HVS 5.1 willing to test.