safepay/sensor.fronius #69: Energy Cost Sensor - last_reset deprecation (2021.9.4→2021.9.5)

thehaxxa, 2021-09-09:
Yet another change in home assistant (_pertaining to energy dashboard_), energy sensor cost has been updated with: 
        `STATE_CLASS_MEASUREMENT` resulting in an warning:

```Entity sensor.fronius_smartmeter_energy_ac_sold_compensation (<class 'homeassistant.components.energy.sensor.EnergyCostSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+energy%22```

See here for details: https://github.com/home-assistant/core/pull/55962

nilrog, 2021-09-09:
This integration is already compatible with the changes in HA 2021.9.x.

We never set both `STATE_CLASS_MEASUREMENT` and `last_reset` at the same time because that was not possible. One of those has been added by you in a custom sensor. I can see that the sensor it is complaining about is `sensor.fronius_smartmeter_energy_ac_sold_compensation` which is not a sensor created by this integration.

Install the latest update and remove your workarounds you have used to get these sensors into the energy panel in HA and all is good.

thehaxxa, 2021-09-09:
> This integration is already compatible with the changes in HA 2021.9.x.
>
> We never set both `STATE_CLASS_MEASUREMENT` and `last_reset` at the same time because that was not possible. One of those has [...]

Please note I already updated to v0.9.7 and was running @colwilliamsnz's fork before the changes were merged.

This is **not** related to the changes in 2021.9, this is a new change made in the point release **2021.9.5**. I can confirm it works fine on 2021.9.4. I'm already running the latest version and I have already removed old configuration in my configuration.yaml. 

The `sensor.fronius_smartmeter_energy_ac_sold_compensation` sensor is generated by Home Assistant in relation to `sensor.fronius_smartmeter_energy_ac_sold` sensor. I have tested reverting back to 2021.9.4 and the warning disappears:
 
Here is my full `configuration.yaml`, (https://paste.debian.net/plain/1211070) I'm confident this relates to PR mentioned.

thehaxxa, 2021-09-09:
Speaking to some developers on Home Assistant Discord, this is an upstream issue that does not require code changes, this is the PR: https://github.com/home-assistant/core/pull/55955 to fix the bug.

It will be added 2021.10. Given the rapid development of energy platform at the moment it seems many changes are occurring.

thehaxxa, 2021-09-09:
@nilrog For reference, this is an upstream bug. Closing Issue.

closed 2021-09-09 by thehaxxa