safepay/sensor.fronius #71: No Day/monthly/year Energy Data from Fronius Symo /Gen24

Harlorad, 2021-09-14:
Hey, im really new in HASS but all things works without the Energy section. 

Here i use this configuartion:

  - platform: fronius_inverter
    name: gen24
    ip_address: 192.168.178.106
    model: gen24
    units: kWh
    powerflow: True
    power_units: kW

and

  - platform: fronius_inverter
    name: symo4.5
    ip_address: 192.168.178.107
    model: symo
    units: kWh

I dont get Data from any Day/Year sensors, here is every value unknown... but i need this for my Energy shard, to use in Solar production and grid configuration...

Have anyone an example for me?

I use HASS  "core-2021.9.6" and the new version from this.

nilrog, 2021-09-14:
You need to turn your questions to Fronius and ask them why this is not available in Gen24. Also the API implementation for the Gen24 seems to be somewhat of a mess.
You can read more about it in #42

Harlorad, 2021-09-15:
> You need to turn your questions to Fronius and ask them why this is not available in Gen24. Also the API implementation for the Gen24 seems to be somewhat of a mess.
> You can read more about it in #42

Damn ok, thx alot i ask Fronius :) why not.

nilrog, 2021-09-16:
I found this document that seems to originate from Fronius. And there they are saying that the Gen24 REST API will be fixed in Q4 2021. But there are no details about what they are "fixing".
[SE_UI_Latest_News_GEN24_EN.pdf](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwjtxuHdi4PzAhXw-ioKHUOgCNEQFnoECCEQAQ&url=http%3A%2F%2F213.33.117.110%2F~%2Fdownloads%2FSolar%2520Energy%2FUser%2520Information%2FSE_UI_Latest_News_GEN24_EN.pdf&usg=AOvVaw1V2ONbk9_R7O7HBkexKUGJ)

RW-7, 2021-09-17:
today i spoke to fronius support about another problem.
but i asked about the api right away, according to support this should be fixed in the upcoming fw.
According to internal information, this should appear on September 29, 2021

Harlorad, 2021-09-17:
> today i spoke to fronius support about another problem.
> but i asked about the api right away, according to support this should be fixed in the upcoming fw.
> According to internal information, this should appear on September 29, 2021

Oh that sounds really great. wonderful. Thx alot. 👍

RW-7, 2021-09-28:
hello, just saw the update online, let's see how it is

RW-7, 2021-09-28:
New functions:
/ Phase accurate feed-in limitation (weakest phase approach) can be set as a separate mode under “Export
Limitation”
/ Solar API adjustments according to API documentation
/ Remote control via Solar Web implemented. In the Web UI under Communication you can now activate the
remote control via Solar Web. This feature gives utilities the possibility to change the feed-in limitation of the
inverter (currently only used in Australia).
/ New error codes for hardware defect in the AC bridge circuit. State 1065 for L1, 1066 for L2 and 1068 for L3
have been added.
/ DC sum current limitation to increase lifetime at certain operating points. (Low DC voltage on PV1, PV2 and
BAT - High DC current, high temperatures).
I_sum= I_PV1+ I_PV2+ I_BAT
1pNext 6kW: I_sum= 39A
1pNext 10kW: I_sum= 46A
3pNext 5kW: I_sum= 30A
3pNext 10kW: I_sum= 50A
Bugfixes:
/ False tripping State 1048 (= AC overcurrent) has been fixed.
/ Neutral wire break detection in emergency power has been fixed. So far no detection in case of error from the
device.
/ No battery management at MPP voltages of 720VDC. At higher MPP voltages it could happen that the battery
has not been charged if first PV and then the BAT was switched on.
/ No battery management if the battery is switched off manually for >10min. during operation and then switched
on again. Battery then started up in energy saving mode. So far only an AC/DC reset helped here.
/ CAN Bus false triggers (=State 1020 & State 1022) fixed.
/ Modbus API via TCP did not work with replacement devices.
/ Load management function - Automatic correction for the "Power surplus" function after a software update to
1.13.9. Depending on the software version, the Consumption and Feed-in parameters were exactly inverted.
(Both on the Web UI and with Modbus/API commands). With the update this is automatically corrected.
/ Web UI function "Measures against FI false trips" did not work.
/ Avoid FI trips when BYD battery is switched off.
/ Wrong error code in case of overload in emergency power. Previously, sometimes only a power too low error
was displayed instead of overload error.
/ State 1143 BugFix. False triggers fixed
/ Cloud Handling BugFix: Devices reconnect automatically to Solar Web after a connection loss.
Changes in country setups:
/ Australia setup change AUS1 - AUS7 due to AS 4777.2
/ Setup Adaption Switzerland (Type A & Type B)
/ New Setup ZA1 for South Africa
/ New setup IE added
/ Setup EE added (Estonia)
/ Setup change Setup SE - FAC values adapted
/ New Setups AUA, AUB and AUC for Australia & Setup NZ1 for New Zealand

kwanm84, 2021-09-29:
I updated to the latest firmware and I removed:

    model: gen24

from my config. The names have all changed again... looks like the old scheme is working and the gen24 setting isn't required anymore. I think everything comes across, but I will do some testing.

Harlorad, 2021-09-29:
Ooooh i go directly online and update my components :D hopes that all works fine, write later 👯

Harlorad, 2021-09-29:
Works well, but Fronius Solar.Api don't give me W from battery :( some ideas?

RW-7, 2021-09-29:
I only get the entire production, no daily or annual production. I have not installed a battery at the moment

nilrog, 2021-09-29:
> I updated to the latest firmware and I removed:
> 
> ```
> model: gen24
> ```
> 
> from my config. The names have all changed again... looks like the old scheme is working and the gen24 setting isn't required anymore (assuming upto date firmware). I think everything comes across, but I will do some testing.
> 
> Edit: Can confirm all the sensors are working properly with the original config (don't need the gen24 with the new firmware). Day and Year total are still null... which were expected looking at the API... and Total energy is now being reported. It has been recording the entire time it seems, as mine started at about 1.62MWh, and according to pvoutput I have done 1.59MWh.

Good that they (finally) fixed the API :)
Looks like we can revert the patch that added "support" for Gen24 soon (and add some notes about this to the readme). Let's just wait a while so that everyone who is using this integration has had time to update their inverters.

Can you just confirm which FW version you installed?

kwanm84, 2021-09-29:
> Can you just confirm which FW version you installed?

Sure.

1.13.13-1

nilrog, 2021-09-29:
I have updated the README with information about the API fixes in Gen24 FW. Let me know if it looks ok and I will create a pull-request.
https://github.com/nilrog/sensor.fronius/commit/b69bf7da7654057cf7d695a364d0500a1d097453

kwanm84, 2021-10-01:
> I have updated the README with information about the API fixes in Gen24 FW. Let me know if it looks ok and I will create a pull-request. [nilrog@b69bf7d](https://github.com/nilrog/sensor.fronius/commit/b69bf7da7654057cf7d695a364d0500a1d097453)

Yea looks good!