safepay/sensor.fronius #79: Gen24 - API + Webserver becomes unavailable

kwanm84, 2021-11-28:
I'm just wondering if anyone else is experiencing this. It seems to be random, and its seems to be happening more often. At some point, the inverters web server becomes inaccessible, and so no API calls can be made. 

Interestingly the solar.web connection still works, so it still has access to the internet and reports stats to that.

It seems to go through stages. It was working perfectly from the 16th to the 22nd (didn't look further back than that)... but on the 22nd at 11:30am it stopped working. It started again by itself at 3:10pm. Stopped at 7:55pm.

23rd - Up @ 1:30am
Down 9:30am
Up 12:15pm
Down 2:30pm
Up 7:15pm
24th - Down 4:30am
Up 8:15am
Down - 9:25am
Up 11:15pm

etc etc.

When I do notice its down, the webserver is inaccessible, I can only assume it fixes itself when the home assistant sensor can get API calls again. 

Now I know this is probably not an issue with this side of things, but I wanted to see if anyone had any similar issues. Maybe the API calls are too much for it and it shits the bed. 

I have considered the wireless network dropping out, but as it was reporting to solar.web the entire time over the last few weeks I figured that wasn't it. Maybe the inverter itself is broken. I have disabled the sensor for now, and will see if it breaks by itself. If it does then I can put it down to the wireless or inverter. I'll try talking directly to fronius once I can narrow down the list of suspects.

Thanks

lochstar, 2021-11-29:
I haven't had this happen. Even when the inverter restarts due to a grid overvoltage issue, the web server is still available. I am connected via ethernet cable though, not wireless.

kwanm84, 2021-11-29:
I have turned it back on... I actually suspect now that it may have been my wifi network. I noticed the last time it died, solar.web was reporting the inverter offline, and hadn't recorded any data... but after it reconnected it uploaded all the missing data... so looked like it was always online when I had checked.

I'm trying much more robust wifi solution now... will see how it goes over the next few days... and if it keeps happening will switch to ethernet.

kwanm84, 2021-11-30:
Much more confident its was a problem with my wireless. I have over 100 devices on my network when all the lights and switches are on.

I know my router was struggling with the smaller devices, they would disconnect randomly (smart lights and switches). I had installed an more enterprise grade access point just for them, but I had the inverter on the normal router wifi. Even though there would have been only a handful of devices left on that wifi, it still must have been struggling with the wifi still or just the shear number of devices. 

Yesterday I replaced it all with better stuff and its been rock solid, if only for 24 hours... but I'll reopen a new one if it comes back and isn't related to the wifi.

closed 2021-11-30 by kwanm84