safepay/sensor.fronius #82: device_state_attributes

Bascht74, 2022-01-07:
I get the following log in HA:

`2022-01-07 11:21:20 WARNING (MainThread) [homeassistant.helpers.entity] Entity sensor.fronius_west_ig_tl_3_6_ac_power (<class 'custom_components.fronius_inverter.sensor.FroniusSensor'>) implements device_state_attributes. Please report it to the custom component author.
`

See: 
http://dev-docs.home-assistant.io/en/master/api/helpers.html

> property device_state_attributes
> Return entity specific state attributes.
> 
> This method is deprecated, platform classes should implement extra_state_attributes instead.

Could you change your integration to the new method?

djsheetu, 2022-02-09:
I am having the same issue for a couple of days as well. Please see if you can update the integration.
Thanks.

Bascht74, 2023-01-09:
no reaction

closed 2023-01-09 by Bascht74